ftctechnh/ftc_app #551: Rev Robotics Laser distance sensor

JimPitcher, 2018-08-26:
Are the configuration file and programming examples available for the Rev Robotics Laser distance sensor?
The MR distance sensor reports a error in hardware.

lucasjman, 2018-08-26:
No it’s all the hardware, the variables and the functions created in other
classes and then imported. If I revert back to android studios 3.4.x it
works fine but if I update to android studios it 4.0 the. It no longer
works. I could be something to do with android studios making compiling
obsolete?

rbrott, 2018-08-26:
The REV Robotics Laser Distance Sensor is not present in the 3.x FTC SDK although support will be added in 4.x (which will hopefully be released soon). If you don't want to wait, you can use the [FGC SDK](https://github.com/FIRST-Global/FGC_2018) which includes a driver (and a [sample](https://github.com/FIRST-Global/FGC_2018/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/SensorREVLaserDistance.java)).

NoahAndrews, 2018-08-26:
You can't really use the FGC SDK except on a Control Hub. Support will come to FTC soon enough.

JimPitcher, 2018-08-27:
Thank you this is what I need.

closed 2018-08-27 by JimPitcher