ftctechnh/ftc_app #557: Build issue with latest code pull

hadronknights, 2018-09-09:
Getting this issue with the latest code pull.

Could not find com.android.tools.lint:lint-gradle:26.1.4.

archishou, 2018-09-09:
not an issue with the ftc-app, https://stackoverflow.com/questions/46972122/could-not-find-com-android-tools-lintlint-gradle-android-studio-3

JBYoshi, 2018-09-14:
I had the same issue. It's not an issue with the app, but it is an issue with the Gradle scripts they provide. Adding this line to the bottom section of `build.common.gradle` fixed my problems (ignore the + sign at the beginning, it's just for formatting).
```diff
 repositories {
     flatDir {
         dirs rootProject.file('libs')
     }
+    google()
 }
 apply from: 'build.release.gradle'
```

I'd like it if this fix could be merged in to the FTC side, but FTC rarely accepts pull requests here.

David10238, 2018-09-14:
I fixed the two Gradle issues by adding google() and mavenCentral() to FtcRobotController's build.gradle repositories block. If FIRST can implement this change or something better, it would help team's by avoiding the issue.

gotpredictions, 2018-09-14:
I grabbed a fresh copy of the code and I was able to build it without any
trouble.  Of course I have nothing in the teamcode :-).  Perhaps it is
because the linters are included in your team code?

hadronknights, 2018-09-14:
yes if google() added to the repository definition in build.common.gradle then the error goes away.  Please add the fix in gradle script before the next update.

dmssargent, 2018-09-14:
@hadronknights @David10238  @JBYoshi 

What is the version of the Android SDK tools locally installed on your computer? You can use the following as a guide: https://developer.android.com/studio/intro/update#sdk-manager.

hadronknights, 2018-09-14:
Here is the sdk tool version :

[image: image.png]

JBYoshi, 2018-09-15:
I have Android SDK Tools 26.1.1 and Android Platform Tools 28.0.1. I also have a bunch of different versions of Android Build Tools. I would expect there to be an update for the SDK tools, but there aren't any updates available for it.

JBYoshi, 2018-09-15:
My teammate had similar issues when updating. I had to add `google()` and `jcenter()` to the repositories section in both `build.common.gradle` and `FTCRobotController/build.gradle` to fix his computer.

mikekochxda, 2018-09-17:
So, for me, it was google(), mavenCentral(), and jcenter().  But why?  There must be a reason this works for other people without having to do this - or only having do to one or two?  Is there a setting somewhere that I (we) have wrong?

repositories {
    flatDir {
        dirs '../libs'
    }
    google()
    mavenCentral()
    jcenter()
}

David10238, 2018-09-17:
I would recommend an interim fix of adding to the repository blocks. I personally know teams some of which have good programmers that have trouble with changing the build files. This would at least be a band-aid until you can implement the offline solution, and more teams would be able to get work done with the latest SDK. I do like that the SDK is completely self-packaged offline. It helps at competitions when I have to upgrade another team's SDK.

JBYoshi, 2018-09-18:
@mikekochxda You shouldn't need `mavenCentral()` if you also have `jcenter()`. According to [the JCenter maintainers](https://jfrog.com/knowledge-base/why-should-i-use-jcenter-over-maven-central/), JCenter has everything Maven Central has.

NoahAndrews, 2018-09-18:
JCenter has (or at least had) a certificate issue recently that I believe is likely to have been the cause of that.