ftctechnh/ftc_app #559: App crashed when detecting controllers

berberman, 2018-09-13:
Hello:
We have a problem that robot controller crashed many times.

[image]

That was the stack trace in logcat. I guess that this exception was caused by incompatible Android 8.0 which we used. And what should we do to solve that? Please help us. 😃

cmacfarl, 2018-09-13:
The call to create a [MediaPlayer](https://developer.android.com/reference/android/media/MediaPlayer#create(android.content.Context,%20android.net.Uri,%20android.view.SurfaceHolder,%20android.media.AudioAttributes,%20int)) failed, and there's no null check on the object upon failure.

Can you please include any portion of the log above this stack trace that is referencing MediaPlayer?

cmacfarl, 2018-09-13:
Also, please provide the make and model of your phones.

SpyGuyIan, 2018-09-13:
I am having the same issue. The Robot Controller crashed when paired with the Driver station, directly after the Mindstorms startup sound. I was testing with 2 personal phones: Zenfone 2 for DS and OnePlus 5T for RC.

Here is a good amount of the log I captured: https://pastebin.com/ssp2UMt3

CSharperMantle, 2018-09-14:
I am having the same issue too. After the Robot Controller has paired, while playing the sound of success, the app crashed. I was using two Redmi-4Xs in China.

berberman, 2018-09-14:
There was no more stack trace about `MediaPlayer` , we were using MI-MIX2S with MIUI9 (customized Android 8) for Robot Controller in China.

gearsincorg, 2018-09-14:
To provide as much information as possible about your phones, can you post a screen grab of the bottom of the Phone's About screen?

rgatkinson, 2018-09-14:
@berberman (and @CSharperMantle) : even if there were no references to MediaPlayer, the context provided by other log entries around the event in question can at times be extremely useful. If you still have the logs available, we'd very much like to see them.

WXY7050, 2018-09-15:
@rgatkinson : Hi, I have the same issue. Here is the logs.
[robotControllerLog.txt](https://github.com/ftctechnh/ftc_app/files/2385629/robotControllerLog.txt). 

@gearsincorg Here is the Phone's About screen.
[image]

Thanks for your help! Looking forward for your reply.

rgatkinson, 2018-09-15:
@WXY7050 : thanks. A common thread here in at least most of these reports is Xiaomi phones. We are actively investigating.

WXY7050, 2018-09-16:
@rgatkinson Thanks for the response. Not all the Xiaomi phones had the Robot Controller APP crashed. Our Redmi 6A works fine, but 5A not. Keep following up.

CSharperMantle, 2018-09-17:
> It seems that we can redefine `SoundPlayingRobotMonitor` and `FtcEventLoopBase` to avoid referring `MediaPlayer`. However, this is only a temporary plan.

@berberman But SoundPlayingRobotMonitor.java doesn't seems to be editable. :-(

berberman, 2018-09-17:
@CSharperMantle Just create your own class, let it to extends `SoundPlayingRobotMonitor`, and override method `playSound(Sound, Int, Consumer<Int>, Runnable)`. Finally, replace `SoundPlayingRobotMonitor` in `createUICallback(UpdateUI)` with your class. You can find my code [there](https://github.com/berberman/ftc_app/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/internal/monitor/MyRobotMonitor.kt). Actually, I haven't test that yet, I'm not sure that it can work. :P

CSharperMantle, 2018-09-19:
@berberman Thanks for your idea! I extended `SoundPlayingRobotMonitor` and it worked well! Here's my code:
```java
package org.firstinspires.ftc.robotcontroller.internal;

import android.support.annotation.Nullable;
import android.support.annotation.RawRes;

import com.qualcomm.ftcrobotcontroller.R;

import org.firstinspires.ftc.ftccommon.external.SoundPlayingRobotMonitor;
import org.firstinspires.ftc.robotcore.external.function.Consumer;

public final class SoundPlayingRobotMonitor_Patched extends SoundPlayingRobotMonitor {
    private int[] _availableSound = {R.raw.nxtstartupsound, R.raw.errormessage, R.raw.warningmessage};

    @Override
    protected void playSound(Sound sound, @RawRes final int resourceId, @Nullable Consumer<Integer> runWhenStarted, @Nullable Runnable runWhenFinished) {
        //Ignored.
        boolean soundMatched = false;
        for (int s: _availableSound) {
            if (resourceId == s)
                soundMatched = true;
        }
        if (soundMatched)
            super.playSound(sound, resourceId, runWhenStarted, runWhenFinished);

    }
}

```

rgatkinson, 2018-09-25:
Today I finally got my hands on a phone which exhibited the crash, a Redmi Note 5A running Android 7.1.2 N2G47HY / MIUI 10 Global 8.8.30 Beta. 

Analysis seems to indicate that the root problem is that on the robot controller phones on which this is failing, the particular sound chimeconnect.wav (see FtcCommon\res\raw) cannot be played by MediaPlayer, though the reason for this is not clear.

This will be fixed in an upcoming release by modifying the sound file slightly to avoid the problem. In the meantime, the bug can be worked around by setting the (public) ``soundConnect`` member of the SoundPlayingRobotMonitor instance referenced by @berberman above to something other than its default R.raw.chimeconnect.