ftctechnh/ftc_app #560: Network status says "unknown" even when connected

FROGbots-4634, 2018-09-14:
It appears that there is a bug in SDK v4.0 which causes the network status to be listed as "unknown" even when it is connected. Have a look at the network status TextView on the RC:

[image]

Steps to reproduce:

1) Open RC app
2) Open DS app
3) Wait for the DS to connect to the RC

SpyGuyIan, 2018-09-15:
We were having a similar issue. The phones were connected through WiFi-Direct in settings and through the "Pair Robot Controller" on the DS, but there was no ping (it was blank) and we also got "Network: unknown, connected". All would be fine if we could still run opmodes but the menus are all blank.

We are running on two Moto G5 devices

cmacfarl, 2018-09-15:
The lack of ping on @SpyGuyIan's phones is not necessarily related to the network status reported here.  The network status indication will be fixed in the next point release.  

@SpyGuyIan, can you please swipe close, or force stop, both the DS and RC, start the RC, and then start the DS?  If they do not pair and you do not get pings please open a new issue and attach both the DS and RC log.

FROGbots-4634, 2020-08-26:
Fixed a long time ago, in SDK v5.1 I believe

closed 2020-08-26 by FROGbots-4634