ftctechnh/ftc_app #563: ability to play audio files on Driver Station

name45, 2018-09-14:
in the release Information it says Added ability to play audio files on Driver Station does that mean from the Robot controller you can play sounds on the driver station?

cmacfarl, 2018-09-15:
Yes.

nrizzio, 2018-09-15:
Is there a special way to play sounds through the Driver Station or do you just use the `SoundPlayer` class as usual?

berberman, 2018-09-16:
While you are using `SoundPlayer#play` to play a sound, the driver station will receive the message as well.

```java
if(isRobotController){
	// Tell the driver station too!
	CommandList.CmdPlaySound cmdPlaySound = new CommandList.CmdPlaySound(msPresentation, soundInfo.hashString, params);
	Command command = new Command(CommandList.CmdPlaySound.Command, cmdPlaySound.serialize());
	command.setTransmissionDeadline(new Deadline(msSoundTransmissionFreshness, TimeUnit.MILLISECONDS));
	NetworkConnectionHandler.getInstance().sendCommand(command);
}
```

> This code snippet lies in `SoundPlayer#startPlayingLoadedSound`.

However, if you want to play sounds on the driver station simply, you can send `CmdPlaySound` to the driver station manually.

Please see class `CommandList$CmdRequestSound` and `SoundPlayer` to get more details.

float23, 2018-09-16:
Do you have sample code for playing a sound in a opmode?

rgatkinson, 2018-09-16:
@float23 : See ``SoundPlayer.getInstance().startPlaying()``. Sounds can live in resources in or files.

@berberman : please avoid using ``CmdPlaySound`` directly; that's logically part of the implementation, and may change from release to release.

float23, 2018-09-18:
for SoundPlayer.getInstance().startPlaying() what do i put for Context?

closed 2020-08-26 by cmacfarl