ftctechnh/ftc_app #566: git clone --depth 1 led to problems?

jdkunzeman, 2018-09-16:
I have struggled to learn git to help the team, but perhaps this will not seem like an issue to accomplished git users.  It is prominently recommended on this site to use the command 

git clone --depth=1 https://github.com/ftctechnh/ftc_app.git

when downloading the app.  I had forked the repository to our team repository a month ago and  created a local copy on a PC with the above command.  At the local level, I added our team code and used git to successfully push the updated app back to our repository.  So far, so good.  

After the new ftc_app was released last week, I used 

git remote add upstream https://github.com/ftctechnh/ftc_app  

to create another remote repository on my local.  I fetched and merged.  There were a few gradle conflicts that I resolved successfully.  

Then I noticed that all five of the classes related to Vuforia in the External Samples section under FTCRobotController were having issues.  However, if I downloaded a zip file of the app from ftctechnh and opened it in Android Studio, these classes did not show any errors.  To resolve the problem with my local I eventually cloned a new local copy from our team's repository without using the --depth =1 modifier.  I then re-added repository ftctechnh/ftc_app, fetched, merged, and resolved the gradle conflicts.  This time there were no problems with the Vuforia classes in External Samples.  I was able to add, commit, and push my updates to our team repository. 

Some questions:
1.  Should the recommendation to use git clone --depth have some qualifying statements added?  
2.  Was there a better way that I could have resolved the issue I saw with the Vuforia classes?  
3.  When the app I pushed to the team repository is pulled to other locals, will they have still have problems since there were set up using git close --depth?  (Somewhat rhetorical, I will this question myself soon enough.)
4.  If so, is there a better method to fix the other locals?  

In general, I feel that it would be great for someone to develop a simple beginner's guide to using git and github with the FTC app and store it in the Wiki.  I have started writing my own guide, but I'm too inexperienced to write one for ftctechnh.  I found the following site to be quite helpful:

https://www.dataschool.io/git-and-github-videos-for-beginners/

JBYoshi, 2018-09-17:
If you use `git fetch --unshallow`, it will undo the effects of `--depth`. I would recommend doing this at home, since it can clog up shared networks.

Although I don't think this should be necessary. What errors show up with Vuforia?

jdkunzeman, 2018-09-18:
I have already fixed all the locals by running git clone without --depth on all the locals.  This solution did appear to fix the problem, but read on.    

I had saved the old app before I rebuilt it.  I've reopened it on two computers, only to find out that the problems I saw are no longer there with that app.  The only actions I have knowingly taken are those that I've already described.  There were not even any computer restarts involved.d  

One example of where I saw a problem from ConceptVuforiaNavigation:
                vuforia = ClassFactory.getInstance().createVuforia(parameters);
The getInstance() portion of the statement was causing an error previously.  I'm not sure what changed to make this class and others  start working.  

I'll close the issue in a couple of days if there's no other input on it.

ftc8424, 2018-09-18:
I have also experienced very strange reactions, to the point of the 
DcMotor class not being found after I do an SDK update over top of an 
existing solution.  Phil Malone suggested closing the project and then 
doing another import on the project (same directory, no changes, just 
getting AS to completely treat it as new) and that's worked every time.  
It worked when I upgraded to 3.5 last year, 3.7 over the summer and now 4.0.

I don't know why it works where things like "Invalidate Caches / 
Restart" don't, but for whatever reason you might want to try just 
closing the project and re-importing it and see if that works around 
strange errors like that.

qwertychouskie, 2018-09-18:
Also try `Build -> Clean Project` (it can also help remove "phantom" OpModes, see https://github.com/ftctechnh/ftc_app/issues/517).

JBYoshi, 2018-09-19:
I use IntelliJ IDEA (which is pretty similar to Android Studio but provides more customization), and when I updated, it told me that the libraries were misconfigured. After removing them and recreating them, everything seemed to work properly. I think you can do the same thing with Android Studio by deleting the `.idea/libraries` directory and then refreshing the Gradle project.

AlecHub, 2018-09-19:
> In general, I feel that it would be great for someone to develop a simple beginner's guide to using git and github with the FTC app and store it in the Wiki. I have started writing my own guide, but I'm too inexperienced to write one for ftctechnh …

Hi @jdkunzeman, a guide would be great however, for most people, it is impossible to adequately learn Git and GitHub without putting it into LOTS and LOTS of practice on a repository or repositories that lend themselves to learning Git and GitHub.  Unfortunately this repository does not lend itself to learning Git and GitHub.  More than likely the amount of code you will add in your fork and the amount of practice you will have in your forks or branches will not suffice either.  I've known software professional who've had sufficient practice on appropriate repositories yet still failed to learn the basics (admittedly, some of those folks were forever corrupted by [SourceSafe](https://en.wikipedia.org/wiki/Microsoft_Visual_SourceSafe) ;)

ibravo, 2018-09-20:
@jdkunzeman , we also struggled with learning/teaching Github, as it has a technical component (ie: how do I merge, branch, clone, rollback, etc) and a usability component (ie: when shall I create a new brach, vs an issue, vs a fork).

Here google and youtube are your best sources of information. I've also put together a short presentation that I give to all new team members that is based on a youtube video that explains how you should be using GitHub as a code repository.

Give it a try: https://docs.google.com/presentation/d/1YBl9Q-NaWTmZSowURRx0vIsAhNFk3QIEQO6O6HqCP80/edit?usp=sharing

jdkunzeman, 2019-09-18:
Helpful comments - thank you.  I have a two page very simple document on Github which I'm attaching.  It's not much, but it could be somewhere to start for those just at the very beginning of their journey.  

[Getting started with GitHub.pdf](https://github.com/ftctechnh/ftc_app/files/3624189/Getting.started.with.GitHub.pdf)

closed 2019-09-18 by jdkunzeman