ftctechnh/ftc_app #567: Feature request:  Download/upload robot configuration (xml) file via web browser

pmichaud, 2018-09-16:
This is a feature request -- it would be awesome to have the option to download (and upload) robot configuration .xml files via the web browser interface.  Currently we can do this by physically connecting the phone and issuing adb push/pull commands, but being able to do it via the web interface would be really good.

Copying the .xml file from one robot controller device to another is very useful for teams that have or use backup robot controllers at an event.  It's also excellent documentation of the device mapping on expansion hubs (e.g. if an expansion hub needs to be swapped on a robot).

If this feature already exists somewhere and I just haven't seen it, then my apologies and gratitude for pointing us in the right direction.

Thanks!

Pm

cmacfarl, 2018-09-16:
If you develop using Android Studio, then configurations may be stored as resources along with the rest of your software.   Drop them in TeamCode/res/xml and when you deploy they will appear on whatever RC you'd deployed to.   Note that AS resource files must be all lowercase.

rgatkinson, 2018-09-16:
In addition to using the ADB command line, in Android Studio, the Device File Explorer (View / Tool Windows / Device File Explorer) will also allow you to view the RC file system. Robot configurations are stored as the .XML files found in /sdcard/FIRST.

pmichaud, 2018-09-16:
Our team has used adb to copy robot configurations -- we connect the phone via USB and then do:

```adb pull /sdcard/FIRST/configname.xml```
or
```adb push configname.xml /sdcard/FIRST```

where "configname" is the name of the robot configuration.  This feature request is looking for a way to do it via the web interface, since we're already able to upload/download .java files and the like.

gearsincorg, 2018-09-17:
Or just plug the phone into your pc, and set the phone’s USB to operate in
media mode.  Then the phone will appear as an external drive.  The config
file will be in the /FIRST folder

>
> --
Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

FROGbots-4634, 2018-09-17:
@gearsincorg wellllllll........... that works except when you create a new one, you have to reboot in order for it to show up on the computer because.... MTP is stupid and........... yeah.

qwertychouskie, 2018-09-17:
MTP is evil :P

gearsincorg, 2018-09-18:
@FROGbots-4634 This works just fine on my ZTE phone.  New configs are available immediately upon creation.  If you have a Motorola phone, you may have to switch out of, and back into, FTP Media mode. 

What is the specific issue you are referring to...   can you provide some helpful information?

@qwertychouskie let's try to avoid noise posts please.  It makes it harder to locate real issues.

FROGbots-4634, 2018-09-18:
@gearsincorg This has been a bug in Android since 2012... Here's an example: https://stackoverflow.com/questions/13737261/nexus-4-not-showing-files-via-mtp

gearsincorg, 2018-09-18:
@FROGbots-4634 That gem of information probably explains why I was having issues with downloading log files from my Moto phones last year.  Luckily transferring config files doesn't have to happen often, so the trick here is "knowing" that you needed a phone reboot for them to be available.

Based on this, it does seems like we currently have several web-based file transfer options (logs, sounds, firmware, camera cal) spread between the General and Blocks world that could be expanded and standardized.

FROGbots-4634, 2018-09-18:
@gearsincorg I mean to be clear, it's not an issue for me, I just always use ADB (wireless ADB actually) so I can push/pull a file from the command line very quickly... it is however an issue for less experienced teams who aren't up to speed with ADB and stuff yet.

FROGbots-4634, 2018-09-18:
@rgatkinson @cmacfarl you might want to look into doing [this](https://stackoverflow.com/a/20413888/6122159) for any files (log/conf/otherwise) saved by the RC.