ftctechnh/ftc_app #570: Issue: Can't Install the RC app on the phone

SoaringWillow, 2018-09-19:
When trying to install the 4.0version of RC and DS updates on our phones from last year, the DS installed just fine, but the RC app file won't install. We tried to connect it to the robot, quit the app, and restart the phone, but none worked. There is a warning on the app  shows is a line of "Warning: unable to open REC Robotics USB Expansion Hub Portal [DQ1687IA]" . And the it simply says "app not installed" when we try to install the update
[image]
[image]

cmacfarl, 2018-09-19:
The bottom phone is running a robot controller whose configuration file, Delta, is looking for an expansion hub whose serial number is DQ1687IA.  I can't tell whether you have it connected to anything.  Is it connected to an expansion hub?  If so, what happens when you try to create a new configuration and scan for hardware.  Does it find anything?  Does it have the same serial number?

The bottom photo shows a phone wherein you've tried to install a robot controller.  Was that intentional or do you want a driver station on the other phone?  If it was intentional can you describe the exact process you used to attempt the installation in as much detail as possible?

gearsincorg, 2018-09-19:
It looks to me like the new app won't install over the old one, and when you run the app, you just get last years app.  You could verify this by using the menu to show the "self inspect" screen.  It will display the app revision.  4.x is this year, 3.x is last year.  You could add a photo of that screen here.

Are you just trying to install the upgrade from the google play store?
(I'm wondering if somone used Android Studio last year to build a new RC App on that phone.  This may prevent the update from working directly from the Google Play store)  

If the new app won't install over the old one, the best thing to do is uninstall the old RC app.
If you use Blocks or OnBotJava, you should download (backup) your opmodes to your PC first.  Do you know how to do this?  

Once you have backed up your programs (and know where they are) you can use the phone's Settings/Apps menu to select and uninstall the FTC_RobotController app.

Then go and re-download and install the app from Google Play Store.

SoaringWillow, 2018-09-21:
After uninstalling and reinstalling the app, we finally managed to update to the latest version of app. Thank you so much.

closed 2018-09-21 by SoaringWillow