ftctechnh/ftc_app #578: ConceptVuforiaNavRoverRuckus.java won't build using OBJ

BayBuilder, 2018-09-25:
I'm trying to build the [ConceptVuforiaNavRoverRuckus.java](https://github.com/ftctechnh/ftc_app/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptVuforiaNavRoverRuckus.java) code sample using On Bot Java and always get this error:

> org/firstinspires/ftc/teamcode/ConceptVuforiaNavRoverRuckus.java line 146, column 31: ERROR: cannot find symbol
> symbol: method getInstance()
> location: class org.firstinspires.ftc.robotcore.external.ClassFactory

The code that generates the error is this:
```
VuforiaLocalizer vuforia;  //Class variable

VuforiaLocalizer.Parameters parameters = new VuforiaLocalizer.Parameters(cameraMonitorViewId);
parameters.vuforiaLicenseKey = VUFORIA_KEY ;
parameters.cameraDirection = CAMERA_CHOICE;
vuforia = ClassFactory.getInstance().createVuforia(parameters);  //Code causing build error

```
The import for ClassFactory is at the top of the .java file, and the documentation shows getInstance() as a method of ClassFactory. All of the other Vuforia samples fail to build with similar errors. Last year I was able to build the ConceptVuMarkIdentification.java sample using the same phones with the previous version of the FTC apps and Android 6.0.1.

Phone: Moto G2 / Moto G Play (similar results on both phones)
FTC App Version: 4.0
Android version: 7.x (recently updated to latest from Verizon)
(from Verizon) Current Software Update v7
Software Version: NPIS26.48-43-2
Android® Security Patch Level: 2018-06-01

FROGbots-4634, 2018-09-25:
I can confirm this as well.

JBYoshi, 2018-09-25:
Have you tried refreshing the Gradle project? There should be an icon in the toolbar to do that.

gearsincorg, 2018-09-25:
I have seen and corrected this problem myself, and as far as I can tell, this error is caused by some remnants of the prior OnBotJava libraries remaining on your Robot Controller phone.  

One effective way to correct this problem is to let the new App completely reset the OnBotJava subsystem on the phone.  There is an "Advanced" setting to permit this on the OnBotJava web interface.

Before making any changes though, I'd download any OnBotJava Opmodes you have on your RC to your computer as a backup.  Simply right click on them (in the tree) and select download...

To reset the OBJ system, follow these directions:

1) Start up the RC app and use your browser to connect to your phone just like you would if you were doing OnBotJava programming.
2) Select OnBotJava, and click on the Settings (GEAR) icon.
3) Scroll down to, and expand the "Advanced Settings" option
4)Click on the RED "RESET" button.

   !!!!!   MAKE sure you have downloaded any OpModes you want to keep BEFORE doing this...!!!!!

   You will be prompted if you really want to do this: Click OK
   And you will be prompted AGAIN... and actually have to type "Yes I want to".  Do it!

Now, if you return to the OnBotJava page, all the opmodes will be gone and you should be able to either re-create them, or restore your saved opmodes by using the Upload Files icon.

Also remember that if you are using Vuforia, you need to replace the dummy Vuforia key with your own key.  See the code comments.

FROGbots-4634, 2018-09-25:
> 
> 
> Have you tried refreshing the Gradle project? There should be an icon in the toolbar to do that.

This is ref OBJ, not AS

BayBuilder, 2018-09-25:
> I have seen and corrected this problem myself, and as far as I can tell, this error is caused by some remnants of the prior OnBotJava libraries remaining on your Robot Controller phone.
>
> One [...]

gearsincorg: Thank you! I did the reset you outlined and the code sample built and ran successfully. This tidbit should be noted in the [Identifying Vuforia VuMarks](https://github.com/ftctechnh/ftc_app/wiki/Identifying-Vuforia-VuMarks) portion of the Wiki - or at least added to the comments in the Vuforia sample code.

closed 2018-09-25 by BayBuilder

dmssargent, 2018-09-26:
This extended bit about the reset being needed is an issue with OnBotJava that is actively being investigated.