ftctechnh/ftc_app #587: Error code on Android Studio

SoaringWillow, 2018-10-05:
The error code is showing below.
[image]

cmacfarl, 2018-10-05:
You've updated to AS 3.2 and google broke the build by not packaging appt2 with 3.2.

Until the Tech Team provides a release that supports 3.2 natively, the solution is to patch this

  https://github.com/ftctechnh/ftc_app/pull/581/files#diff-c197962302397baf3a4cc36463dce5ea

into your build.gradle file.

TigranTT, 2018-10-06:
change in "build.gradle" and add highlighted. 
[image]

 'compile' is obsolete and has been replaced with 'implementation' and 'api'. 
change it in "build.release.gradle"
[image]

cmacfarl, 2018-10-06:
> ![screen shot 2018-10-05 at 11 05 29
> ![screen shot 2018-10-05 at 11 06 09

The implementation change is in the alpha on ftctechnh/ftc_app.  It's
not strictly needed to fix the build break referenced.   The
repositories block addition is needed to fix the build break related
to google's moving around of dependencies in the 3.2 build tools.

cmacfarl, 2018-11-03:
Fixed with 4.3 by adding the external repositories to the dependencies block.

closed 2018-11-03 by cmacfarl