ftctechnh/ftc_app #592: core power distribution modue(CPDM)

qmmp123, 2018-10-10:
Hi, my team has issue with scanning usb devices. We use motorola G3. We connected 3 motor controllers and servo controller to CPDM but in list of devices we found only 1 device which we connected to port with large index. But there is no problem with zte speed (n9130
[robotControllerLog[1].txt](https://github.com/ftctechnh/ftc_app/files/2464079/robotControllerLog.1.txt)
)

gearsincorg, 2018-10-10:
I've not seen a G3.  What Android OS is it using?

qmmp123, 2018-10-10:
Android 6.0

gearsincorg, 2018-10-10:
OK, so 6.0 should be OK.

This is what I would check:

1) All CORE controllers are lit up, and the CPDU has 4 LEDs showing.
  If there are only 3, then a voltage regulator is bad, OR the USB port is not initializing.
2) Plug a USB cable from the CPDU to your PC and run the Modern Robotics Discovery too to ensure that all the controllers are responding, and check the firmware versions.
3) Is the SAME RC App running on the ZTE as the G3 (same SDK version)

qmmp123, 2018-10-11:
1) All ok
2) All ok
3) All ok
Any ideas?

gearsincorg, 2018-10-11:
I'm stumped.   The log file clearly shows that the app knows the modules should be out there, but it can't find them.

This is what your attempts to scan the bus is showing, right?  Just one device?
Has the G3 ever worked before?
Are you using last year's app rev 3.x or this year's 4.x
If you are using 4.x, is there any way you can build and deploy 3.x on that phone as a test?

qwertychouskie, 2018-10-11:
When you say G3 I assume you mean the "Motorola Moto G 3rd Generation"?

qmmp123, 2018-10-12:
Yep

qmmp123, 2018-10-15:
So downgrading did not help