ftctechnh/ftc_app #598: REV Firmware PIDF Implementation

rbrott, 2018-10-15:
REV ExH firmware v1.8.2 introduced the new PIDF DC motor control mode. Given that there is little documentation regarding its implementation, I was hoping someone could clear up the following questions. For simplicity, let's neglect positional control and focus solely on closed-loop velocity control.

1. The "F" term appears to be a velocity feedforward term which acts as a multiplier to scale the provided power value to a setpoint. More specifically, the setpoint is determined as `setpoint = kV * power` where `kV` is the "F" gain. Is this correct?

1. What are the units of `kV`/the "F" gain?

1. The docs seem to imply that `kV = 0` most closely matches the legacy PID functionality. Does this suggest that the legacy PID setpoint was always zero? Or would `kV = 1` be a closer match?

1. Are there any fundamental differences between the PIDF and legacy PID modes aside from the addition of the feedforward gain?

1. How does this relate to "improved DC motor output linearity"? Is this linearity procedure applied in all control modes? What is the linearization procedure?

@rgatkinson

AustinShalit, 2018-10-15:
1. The formula you are looking for is PowerOut = (Other PID terms) + kV * Setpoint
2. Gain in PID(F) controllers is unitless.
3. No.  See my answer to number 1.  The F term adds something else to the output.  If we set it to zero, the output ends up being just the other PID terms.
4. Yes.  Legacy PID had some weird special cases to help improve control but these were not documented.  I recommend PIDF for the most consistent experience.
5. PIDF relies on linear output.  So the change would not be possible without the linear output change.  This change is applied in all control modes.  The change is a configuration change of the signal we are sending to the motor.  There is no math or formula changing the output to make it linear.

rbrott, 2018-10-15:
2. If the output is voltage and the input/original setpoint is velocity, doesn't that imply that `kV` has units?

4. Are there any legacy PID special cases that could've resulted in this response: https://i.imgur.com/P13gobG.png. This is with REV ExHs, default PID coefficients, 20:1 Orbital NR belt-driven mecanum drive, and a ~40lb robot. Ignoring the phase gain, there are strange oscillations right before the velocity goes to zero. As far as we could tell, this was not a mechanical issue (and the open-loop response lacks these oscillations), and it appeared quite consistently over a wide variety of tunings.

On a somewhat-related note, what is the reasoning behind FIRST/DEKA/REV not releasing the firmware sources?

I appreciate your prompt response.

rgatkinson, 2018-10-15:
PID/PIDF systems indeed work better if the response of the measured variable (here, velocity) is linear in the control variable (here, PWM duty cycle). But I think it's a bit strong to state that such linearity is a strict necessity.

In previous versions of the firmware, the PWM frame rate was 1KHz; it is now 10Khz (for all control modes, as Austin mentioned). That this increases linearity of response is not at all obvious, but it does. A place to start pulling on this thread further, if one is interested is [here](https://github.com/rgatkinson/RobotPhysics/blob/master/RippleCurrent.pdf).

closed 2018-10-17 by rbrott