ftctechnh/ftc_app #601: IMU Initialization Error

blah151, 2018-10-17:
Hi,

When initializing our robot using the REV hub's integrated imu, we often receive the following error:  "Warning: problem with imu."  Our team, and many others, have experienced this issue multiple times from the beginning of Relic Recovery until the present; see the following thread on the forum (which has not been resolved):  https://ftcforum.usfirst.org/forum/first-tech-challenge-community-forum-this-is-an-open-forum/teams-helping-teams-programming/60585-imu-initialization-error.  Here is a list of the steps required to reproduce the error.

1.  In the configuration, add an I2c device to port 0 called "imu" (which should be already there).  Remove the automatically added imu from the second REV hub.
2.  Add the following code to initialize the imu:
        BNO055IMU.Parameters parameters = new BNO055IMU.Parameters();
        parameters.angleUnit = BNO055IMU.AngleUnit.DEGREES;
        parameters.accelUnit = BNO055IMU.AccelUnit.METERS_PERSEC_PERSEC;
        parameters.calibrationDataFile = "AdafruitIMUCalibration.json";
        parameters.loggingEnabled = true;
        parameters.loggingTag = "IMU";
        imu = hardwareMap.get(BNO055IMU.class, "imu");
        imu.initialize(parameters);

3.  Place initialization code before waitForStart().  In the forum, it was suggested that initializing after waitForStart() might solve the problem; I do not know if this would, but from the discussion it seems that initializing after waitForStart() only makes the problem less common.

4.  Initialize robot.  After waiting a few seconds, start the robot.  The driver station should say, "Warning: problem with imu."

While the driver station displays this error, the imu will of course fail to operate properly.  An attempt to use it to turn to a specified angle will, in most cases, cause the robot to spin continuously.  
I have attached a robot controller log file including the time at which this error occurred.  The relevant time is 2018-10-16 20:33:07.

[Imu-Error_Log.txt](https://github.com/ftctechnh/ftc_app/files/2488325/Imu-Error_Log.txt)

vwestin, 2018-10-17:
If you do the proper initialization of the IMU in the init() routine this should go away. Unless you do it in init() you are likely to have trouble. Our teams to it just after we map the hardware, just like the example code does. 

                                -Vince

Vince Westin
404-831-0012
http://www.linkedin.com/in/vincewestin
(mobile mail - please excuse typos)

"In a nation that is in servitude, one works more to preserve than to acquire; in a free nation, one works more to acquire than to preserve" - Charles de Montesquieu, The Spirit of the Laws

ftc10179ClassUser, 2020-01-20:
Ok, so our team uses this weird setup, but in the end, our IMU is initialized in the init() function call. Why  does it not work? Also, what causes the IMU to fail to initialize outside of init()? How can we fix it?

ftc10179ClassUser, 2020-01-20:
> If you do the proper initialization of the IMU in the init() routine this should go away. Unless you do it in init() you are likely to have trouble. Our teams to it just after we map the hardware, [...]

Ok, so our team uses this weird setup, but in the end, our IMU is initialized in the init() function call. Why does it not work? Also, what causes the IMU to fail to initialize outside of init()? How can we fix it?

gearsincorg, 2020-01-20:
What Weird Setup?  Please be specific.

Are you including this line?
parameters.calibrationDataFile = "AdafruitIMUCalibration.json";

This is telling the gyro code to read a calibration file.
Have you created this file?  
It comes from running the SensorBNO055IMUCalibration sample opmode.

If you don't have a calibration file, I'd suggest removing this line of code.

markhaye, 2020-05-20:
I am also seeing this error intermittently.  I am using a REV Control Hub.  My init code looks like this:

    // method to initialize the hardware
    public static void init(LinearOpMode _myOpMode)
    {
        // save reference to opMode
        myOpMode = _myOpMode;

        BNO055IMU.Parameters parameters = new BNO055IMU.Parameters();
        parameters.angleUnit           = BNO055IMU.AngleUnit.DEGREES;
        parameters.accelUnit           = BNO055IMU.AccelUnit.METERS_PERSEC_PERSEC;
        // haven't written a calibration file yet
        // parameters.calibrationDataFile = "AdafruitIMUCalibration.json"; // see the calibration sample opmode
        parameters.loggingEnabled      = false;

        // set up internal gyro sensor
        imu = myOpMode.hardwareMap.get(BNO055IMU.class, "imu"); // was Gyroscope.class
        imu.initialize(parameters);
...

All opmodes call this same init routine.  Sometimes it works, sometimes it reports the error.  Sometimes it reports the same error for the REV color/range sensor (that’s the only other device on the I2C bus).  Sometimes restarting the robot helps, sometimes not.

I have tried changing the imu's device type in the config to be "Adafruit IMU" instead of "REV Expansion Hub IMU", but no change in behavior.