ftctechnh/ftc_app #609: Method getInstance() not found in ClassFactory

cwongmath, 2018-10-22:
The method getInstance() isn't found in the class ClassFactory. It always returns a build error: 
[image]
The error happens in any Vuforia sample files.

JBYoshi, 2018-10-23:
Try refreshing your Gradle project.

dmssargent, 2018-10-24:
@JBYoshi Refreshing the Gradle project is rather useless as OnBotJava doesn't interact with Gradle in this context.

gearsincorg, 2018-10-24:
This error is caused because an older OnBotJava library is hanging around on your RC phone.

One effective way to correct this problem is to let the new App completely reset the OnBotJava subsystem on the phone. There is an "Advanced" setting to permit this on the OnBotJava web interface.

Before making any changes though, I'd download any OnBotJava Opmodes you have on your RC to your computer as a backup. Simply right click on them (in the tree) and select download...

To reset the OBJ system, follow these directions:

!!!!! MAKE sure you have downloaded any OpModes you want to keep, BEFORE doing this...!!!!!

1. Start up the RC app and use your browser to connect to your phone just like you would if you were doing OnBotJava programming.
2. Select OnBotJava, and click on the Settings (GEAR) icon.
3. Scroll down to, and expand the "Advanced Settings" option
4. Click on the RED "RESET" button.
5. You will be prompted if you really want to do this: Click OK 
   And you will be prompted AGAIN... and actually have to type "Yes I want to". Do it!

Now, if you return to the OnBotJava page, all the opmodes will be gone and you should be able to either re-create them, or restore your saved opmodes by using the Upload Files icon.

cwongmath, 2018-10-25:
The latest comment worked. Thanks so much!

closed 2018-10-25 by cwongmath