ftctechnh/ftc_app #614: Tensorflow Opmode Crashes when Run

brandonwang1, 2018-10-31:
The tensorflow sample opmode (ConceptTensorFlowObjectDetection) crashes when run.
Relevant logs are here:
https://gist.github.com/brandonwang1/f0f166e50be4506999df9d7023c58c95

It looks like the org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation class imports a nonexistent class com.google.ftcresearch.tfod.util.Size which causes the crash.

FROGbots-4634, 2018-10-31:
Same issue here...

float23, 2018-10-31:
I havent looked at the log but it is crashing fir me to

ItchyCatIII, 2018-10-31:
I have the same issue using their TensorFlow op mode; the package "com.google.ftcresearch" doesn't appear to exist, and I can't edit the file that imports it.  When it runs the import, the FTC Robot Controller app crashes completely.

closed 2018-10-31 by ftctechnh

ItchyCatIII, 2018-10-31:
I just downloaded this again, and I'm getting the same thing.  The `import com.google.ftcresearch.tfod.util.Size` is still there and can't be found, causing the FTC Robot Controller app to crash (from logcat, the error occurred trying to find that class).

Here's the exact text of my error from logcat, in case it helps:

```
10-31 16:03:36.637 1214-1405/? E/AndroidRuntime: FATAL EXCEPTION: LinearOpMode main
    Process: com.qualcomm.ftcrobotcontroller, PID: 1214
    java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/ftcresearch/tfod/util/Size;
        at org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.<init>(CameraInformation.java:38)
        at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.<init>(VuforiaFrameGenerator.java:63)
        at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.<init>(TFObjectDetectorImpl.java:121)
        at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:87)
        at org.firstinspires.ftc.teamcode.GOFNeuralVisionTest.initTfod(GOFNeuralVisionTest.java:116)
        at org.firstinspires.ftc.teamcode.GOFNeuralVisionTest.runOpMode(GOFNeuralVisionTest.java:33)
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
        at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
        at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
        at java.lang.Thread.run(Thread.java:818)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "com.google.ftcresearch.tfod.util.Size" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
        at org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.<init>(CameraInformation.java:38) 
        at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.<init>(VuforiaFrameGenerator.java:63) 
        at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.<init>(TFObjectDetectorImpl.java:121) 
        at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:87) 
        at org.firstinspires.ftc.teamcode.GOFNeuralVisionTest.initTfod(GOFNeuralVisionTest.java:116) 
        at org.firstinspires.ftc.teamcode.GOFNeuralVisionTest.runOpMode(GOFNeuralVisionTest.java:33) 
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305) 
        at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737) 
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
        at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793) 
        at java.lang.Thread.run(Thread.java:818) 
    	Suppressed: java.lang.ClassNotFoundException: com.google.ftcresearch.tfod.util.Size
        at java.lang.Class.classForName(Native Method)
        at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
        at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
        		... 14 more
     Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
10-31 16:06:49.078 1934-2114/? E/AndroidRuntime: FATAL EXCEPTION: LinearOpMode main
    Process: com.qualcomm.ftcrobotcontroller, PID: 1934
    java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/ftcresearch/tfod/util/Size;
        at org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.<init>(CameraInformation.java:38)
        at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.<init>(VuforiaFrameGenerator.java:63)
        at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.<init>(TFObjectDetectorImpl.java:121)
        at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:87)
        at org.firstinspires.ftc.teamcode.GOFNeuralVisionTest.initTfod(GOFNeuralVisionTest.java:116)
        at org.firstinspires.ftc.teamcode.GOFNeuralVisionTest.runOpMode(GOFNeuralVisionTest.java:33)
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
        at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300)
[... cut ...]
        at java.lang.Thread.run(Thread.java:818) 
    	Suppressed: java.lang.ClassNotFoundException: com.google.ftcresearch.tfod.util.Size
        at java.lang.Class.classForName(Native Method)
        at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
        at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
        		... 14 more
     Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
```

qwertychouskie, 2018-11-01:
The fix is now released in 4.3.