ftctechnh/ftc_app #615: Moto E4 phones disconnect when init-ed waiting for play. May be a DS-only issue?

cheer4ftc, 2018-10-31:
See this thread for details.  https://ftcforum.usfirst.org/forum/ftc-technology/67146-moto-e4-phones-disconnecting

David10238, 2018-11-15:
I am also having the problems. We are using TensorFlow, so that is Vuforia frame based.

cheer4ftc, 2018-11-15:
For teams having this problem (@David10238 ), there is a temporary workaround.
The workaround is: make sure your robot is sending back telemetry messages during the extended initialization phase.

In an iterative opMode, this means send back a telemetry message in the init_loop() method, for example `telemetry.addData("Status", "Waiting in Init");`

In a linearOpMode, this may mean replacing `waitForStart();` (or a comparable waiting loop before start is pressed) with something like the following:
`while (!opModeIsActive()&&!isStopRequested()) {     
telemetry.addData("Status", "Waiting in Init");     
telemetry.update(); 
}`