ftctechnh/ftc_app #620: TensorFlow Concept example issue

sashajeltuhin, 2018-11-01:
Trying to incorporate the example with TensorFlow object tracking. The app crashes with the error below. Any suggestions?

Phone Motorola G Play
Android 7.1.1
Robot Controller v 4.2

Error:
RobotCore: thread: 'LinearOpMode main' starting...
10-31 16:17:28.841 19742 20134 V Robocol : sending CMD_NOTIFY_INIT_OP_MODE(506), attempt: 0
10-31 16:17:29.059 19742 20199 V CameraManager: construct(hash=0x0ddfbce3|inst#=0)
10-31 16:17:29.083 19742 20199 D Uvc     : [jni.cpp:72] JNI_OnLoad()...
10-31 16:17:29.082 20199 20199 W com.qualcomm.ftcrobotcontroller: type=1400 audit(0.0:255): avc: denied { read } for uid=10125 comm=4C696E6561724F704D6F6465206D61 name="devices" dev="sysfs" ino=5517 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=0
10-31 16:17:29.083 19742 20199 D Uvc     : [JniEnv.h:32] ScopedJniEnv::onJniLoad()...
10-31 16:17:29.083 19742 20199 D Uvc     : [JniEnv.h:32] ...ScopedJniEnv::onJniLoad()
10-31 16:17:29.083 19742 20199 D Uvc     : [jni.cpp:72] ...JNI_OnLoad()
10-31 16:17:29.083 19742 20199 D libusb  : [ 0.000200] [00004ee7] libusb: debug [isSysfsAvailable] opendir(/sys/bus/usb/devices) failed
10-31 16:17:29.084 19742 20199 W libusb  : [ 0.000283] [00004ee7] libusb: warning [op_init] sysfs(/sys/bus/usb/devices) unavailable
10-31 16:17:29.088 19742 20199 V UvcContext: construct(pointer=0x96471000|inst#=0)
10-31 16:17:29.091 19742 20199 D Vuforia : using camera by direction (classic): BACK
10-31 16:17:29.273 19742 20199 I System.out: Native library libVuforia.so loaded
10-31 16:17:29.274 19742 20199 D Vuforia : Vuforia.init()...
10-31 16:17:29.289 19742 20199 I AR      : Vuforia selected rendering API OpenGL ES 2.x
10-31 16:17:29.289 19742 20199 I AR      : Vuforia SDK version 7.4.3
10-31 16:17:29.335 19742 20199 I CameraManagerGlobal: Connecting to camera service
10-31 16:17:29.783 19742 20199 D Vuforia : ...Vuforia.init()
10-31 16:17:29.806 19742 20208 D NetworkSecurityConfig: No Network Security Config specified, using platform default
10-31 16:17:29.918 20224 20224 W Thread-32: type=1400 audit(0.0:256): avc: denied { read } for uid=10125 name="gpuclk" dev="sysfs" ino=11560 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=file permissive=0
10-31 16:17:29.921 19742 20224 W Adreno-ES20: <get_gpu_clk:229>: open failed: errno 13
10-31 16:17:30.176 19742 20199 I AR      : ObjectTracker: This device doesn't support model targets extended tracking, extended tracking will be turned off
10-31 16:17:30.750 19742 20199 V RobotCore: thread: ...terminating 'LinearOpMode main'
10-31 16:17:30.750 19742 20199 E ThreadPool: exception thrown in thread pool; ignored
10-31 16:17:30.753 19742 20199 E ThreadPool: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/ftcresearch/tfod/util/Size;
10-31 16:17:30.755 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.<init>(CameraInformation.java:38)
10-31 16:17:30.757 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.<init>(VuforiaFrameGenerator.java:63)
10-31 16:17:30.758 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.<init>(TFObjectDetectorImpl.java:121)
10-31 16:17:30.760 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:87)
10-31 16:17:30.761 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.teamcode.skills.MineralDetection.initTfod(MineralDetection.java:225)
10-31 16:17:30.763 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.teamcode.skills.MineralDetection.init(MineralDetection.java:113)
10-31 16:17:30.764 19742 20199 E ThreadPool: 	at org.firstinspires.ftc.teamcode.OpModes.CameraTestMode.runOpMode(CameraTestMode.java:85)
10-31 16:17:30.766 19742 20199 E ThreadPool: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
10-31 16:17:30.769 19742 20199 E ThreadPool: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
10-31 16:17:30.772 19742 20199 E ThreadPool: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300)
10-31 16:17:30.774 19742 20199 E ThreadPool: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
10-31 16:17:30.776 19742 20199 E ThreadPool: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
10-31 16:17:30.777 19742 20199 E ThreadPool: 	at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
10-31 16:17:30.778 19742 20199 E ThreadPool: 	at java.lang.Thread.run(Thread.java:761)
10-31 16:17:30.783 19742 20199 E ThreadPool: Caused by: java.lang.ClassNotFoundException: Didn't find class "com.google.ftcresearch.tfod.util.Size" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm, /system/fake-libs, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]
10-31 16:17:30.784 19742 20199 E ThreadPool: 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
10-31 16:17:30.785 19742 20199 E ThreadPool: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
10-31 16:17:30.785 19742 20199 E ThreadPool: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
10-31 16:17:30.786 19742 20199 E ThreadPool: 	... 14 more
--------- beginning of crash
10-31 16:17:30.790 19742 20199 E AndroidRuntime: FATAL EXCEPTION: LinearOpMode main
10-31 16:17:30.790 19742 20199 E AndroidRuntime: Process: com.qualcomm.ftcrobotcontroller, PID: 19742
10-31 16:17:30.790 19742 20199 E AndroidRuntime: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/ftcresearch/tfod/util/Size;
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.<init>(CameraInformation.java:38)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.<init>(VuforiaFrameGenerator.java:63)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.<init>(TFObjectDetectorImpl.java:121)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:87)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.teamcode.skills.MineralDetection.initTfod(MineralDetection.java:225)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.teamcode.skills.MineralDetection.init(MineralDetection.java:113)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at org.firstinspires.ftc.teamcode.OpModes.CameraTestMode.runOpMode(CameraTestMode.java:85)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:761)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: Caused by: java.lang.ClassNotFoundException: Didn't find class "com.google.ftcresearch.tfod.util.Size" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm, /system/fake-libs, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
10-31 16:17:30.790 19742 20199 E AndroidRuntime: 	... 14 more
10-31 16:17:30.923 19742 19742 D AR      : RenderManager: Could not retrieve a valid GLSurfaceView in view hierarchy, therefore cannot set any render mode

NoahW314, 2018-11-01:
Upgrade the app to version 4.3.  It fixes this issue.

cmacfarl, 2018-11-01:
> org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.(CameraInformation.java:38)
> org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.(VuforiaFrameGenerator.java:63)
> org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.(TFObjectDetectorImpl.java:121)

Update to 4.3.   It fixes the TF crash.

ftcteam8045, 2018-11-01:
I just merged to 4.3 from 4.0,  Android Studio 3.0,  the libraries still come up as unresolved?  
import org.firstinspires.ftc.robotcore.external.tfod.TFObjectDetector;
import org.firstinspires.ftc.robotcore.external.tfod.Recognition;

I'll try downloading the zip file...

sashajeltuhin, 2018-11-01:
4.3 zip worked for me

gorpong, 2018-11-01:
Try FIle > Invalidate Caches / Restart in Android Studio.  Once I did that, everything was found.

An alternative is to close the project and then do another 'Import' of the same project and same directory.  AS treats it as a new project and doesn't use any cached information from it.

cmacfarl, 2018-11-01:
+1 Invalidate caches.

Android Studio can get confused when you add new dependencies.

ftcteam8045, 2018-11-01:
Thanks, yes the invalidate caches worked, Hadn't run into that before.

ftcteam8045, 2018-11-01:
Invalidating caches/Restart did NOT fix my problem,  but closing and re-importing the local project did.  Thanks for the help.

ftcteam8045, 2018-11-01:
thanks, the reimport did the trick.

closed 2018-11-03 by cmacfarl