ftctechnh/ftc_app #621: Repeated tapping of the 'init' button causes the command to be sent to the RC again

FROGbots-4634, 2018-11-01:
Over the years, dozens of users have had strange problems when "double tapping" the init button on the DS - and I think I've figured out why.

**If after the initial press, the 'init' button on the DS is pressed again before the ack packet from the RC is processed, the DS sends *another* init command to the RC.** This causes the SDK to interrupt the currently running init code, start and immediately stop the opmode, and then start the init code again.

Have a look at this log from me tapping the init button twice as fast as possible:

```
11-01 00:16:58.504 14243-14339/com.qualcomm.ftcrobotcontroller V/Robocol: received command: CMD_INIT_OP_MODE(11948) InitTest
11-01 00:16:58.508 14243-14338/com.qualcomm.ftcrobotcontroller I/RobotCore: ******************** START - OPMODE InitTest ********************
11-01 00:16:58.511 14243-14350/com.qualcomm.ftcrobotcontroller I/RobotCore: Attempting to switch to op mode InitTest
11-01 00:16:58.512 14243-14350/com.qualcomm.ftcrobotcontroller V/ThreadPool: container(0x0cb3e108: LinearOpMode) added id=428 TID=0 count=1
11-01 00:16:58.516 14243-14646/com.qualcomm.ftcrobotcontroller V/RobotCore: thread: 'LinearOpMode main' starting...
11-01 00:16:58.516 14243-14646/com.qualcomm.ftcrobotcontroller I/System.out: RUNNING OPMODEINIT
11-01 00:16:58.523 14243-14371/com.qualcomm.ftcrobotcontroller V/Robocol: sending CMD_NOTIFY_INIT_OP_MODE(2176), attempt: 0
11-01 00:16:58.608 14243-14339/com.qualcomm.ftcrobotcontroller V/Robocol: received command: CMD_SET_MATCH_NUMBER(11958) 0
11-01 00:16:58.609 14243-14339/com.qualcomm.ftcrobotcontroller V/Robocol: received command: CMD_INIT_OP_MODE(11959) InitTest
11-01 00:16:58.613 14243-14338/com.qualcomm.ftcrobotcontroller I/RobotCore: ******************** START - OPMODE InitTest ********************
11-01 00:16:58.614 14243-14350/com.qualcomm.ftcrobotcontroller V/ThreadPool: waiting for service user linear op mode
11-01 00:16:59.266 14243-14646/com.qualcomm.ftcrobotcontroller I/System.out: END
11-01 00:16:59.266 14243-14646/com.qualcomm.ftcrobotcontroller V/RobotCore: thread: ...terminating 'LinearOpMode main'
11-01 00:16:59.266 14243-14350/com.qualcomm.ftcrobotcontroller V/ThreadPool: service user linear op mode terminated in awaitTermination()
11-01 00:16:59.266 14243-14646/com.qualcomm.ftcrobotcontroller V/ThreadPool: container(0x0cb3e108: LinearOpMode) removed id=428 TID=14646 count=1
11-01 00:16:59.266 14243-14350/com.qualcomm.ftcrobotcontroller V/ThreadPool: executive service user linear op mode(0x0cb3e108) is terminated
11-01 00:16:59.268 14243-14350/com.qualcomm.ftcrobotcontroller I/RobotCore: Attempting to switch to op mode InitTest
11-01 00:16:59.269 14243-14350/com.qualcomm.ftcrobotcontroller V/ThreadPool: container(0x061232c6: LinearOpMode) added id=429 TID=0 count=1
11-01 00:16:59.270 14243-14653/com.qualcomm.ftcrobotcontroller V/RobotCore: thread: 'LinearOpMode main' starting...
11-01 00:16:59.270 14243-14653/com.qualcomm.ftcrobotcontroller I/System.out: RUNNING OPMODEINIT
11-01 00:16:59.282 14243-14371/com.qualcomm.ftcrobotcontroller V/Robocol: sending CMD_NOTIFY_INIT_OP_MODE(2193), attempt: 0
```
Notice that the first init cmd is received at `16:58.504` and the second at `16:58.609`.

For reference, here was the opmode that was used:

```
@TeleOp
public class InitTest extends LinearOpMode
{
    @Override
    public void runOpMode()
    {
        System.out.println("RUNNING OPMODEINIT");
        long time = System.currentTimeMillis();
        while (System.currentTimeMillis() - time < 750);
        waitForStart();
        while (opModeIsActive());
        System.out.println("END");
    }
}
```
Since commands are repeated until they're acknowledged, there should be no need to re-send the init command. Subsequent presses of the init button should simply be ignored.

gearsincorg, 2018-11-01:
It seems like duplicate Init actions should be ignored at both ends.  RD and RC.

By extension, I wonder if a similar condition could be happening on duplicate START presses (possibly missing the STOP condition).  I'll make sure this is tagged to the Tech Team.

gearsincorg, 2019-06-15:
This problem is resolved in the upcoming 5.0 rev.  
Duplicate button presses are ignored on Driver Station.

closed 2019-06-15 by gearsincorg

FROGbots-4634, 2019-06-16:
@gearsincorg awesome, and thanks for the follow-up too :)