ftctechnh/ftc_app #622: Tensorflow sample in Blocks gives error on ZTE Speed phone, app v4.3

slylockfox, 2018-11-01:
Using Blocks on ZTE phone with app v4.3, using the ConceptTensorFlowObjectDetection sample, I get this error a few seconds after clicking Init:  Fatal error occurred while executing the block labeled "call Tfod.initialize".  I'll look at the log in a followup.

slylockfox, 2018-11-01:
Looks like perhaps the ZTE phones will not support it.  From the log:

    11-01 12:56:17.261  5618  5711 D Vuforia : Vuforia.init()...
    11-01 12:56:17.271  5618  5711 D SensorManager: com.qualcomm.ftcrobotcontroller --- call SystemSensorManager
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: light, handle=0
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: lis3dh-accel, handle=1
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: akm09911-mag, handle=2
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: proximity, handle=3
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: oem-orientation, handle=4
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: oem-linear-acceleration, handle=5
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: oem-rotation-vector, handle=6
    11-01 12:56:17.271  5618  5711 D SensorManager: found sensor: oem-gravity, handle=7
    11-01 12:56:17.271  5618  5711 I AR      : Vuforia selected rendering API OpenGL ES 2.x
    11-01 12:56:17.271  5618  5711 I AR      : Vuforia SDK version 7.4.3
    11-01 12:56:18.251  5618  5711 D Vuforia : ...Vuforia.init()
    11-01 12:56:18.741  5618  5711 I AR      : ObjectTracker: This device doesn't support model targets extended tracking, extended tracking will be turned off
    11-01 12:56:19.471  5618  5703 I AR      : ObjectTracker: Successfully created dataset
    11-01 12:56:19.471  5618  5703 D Vuforia : loading data set 'RoverRuckus'...
    11-01 12:56:19.941  5618  5703 D Vuforia : ... done loading data set 'RoverRuckus'
    11-01 12:56:19.951  5618  5703 W System.err: java.lang.RuntimeException: This Android device is not compatible with Tensor Flow Object Detection.
    11-01 12:56:19.961  5618  5703 W System.err: 	at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:90)
    11-01 12:56:19.961  5618  5703 W System.err: 	at org.firstinspires.ftc.robotcore.external.tfod.TfodBase.initialize(TfodBase.java:55)
    11-01 12:56:19.961  5618  5703 W System.err: 	at com.google.blocks.ftcrobotcontroller.runtime.TfodBaseAccess.initialize(TfodBaseAccess.java:58)
    11-01 12:56:19.961  5618  5703 W System.err: 	at com.android.org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
    11-01 12:56:19.961  5618  5703 W System.err: 	at com.android.org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:24)
    11-01 12:56:19.961  5618  5703 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:102)
    11-01 12:56:19.961  5618  5703 W System.err: 	at android.os.Looper.loop(Looper.java:136)
    11-01 12:56:19.961  5618  5703 W System.err: 	at android.os.HandlerThread.run(HandlerThread.java:61)
    11-01 12:56:19.971  5618  5703 E RobotCore: BlocksOpMode - "Tf-Object-Detect" - main/JavaBridge - caughtException - message is Error: Error calling method on NPObject.
    11-01 12:56:19.971  5618  5703 I RobotCore: BlocksOpMode - "Tf-Object-Detect" - main/JavaBridge - scriptFinished
    11-01 12:56:19.971  5618  5699 I RobotCore: BlocksOpMode - "Tf-Object-Detect" - main/LinearOpMode main - runOpMode - after while !scriptFinished loop
    11-01 12:56:19.971  5618  5699 I RobotCore: BlocksOpMode - "Tf-Object-Detect" - main/LinearOpMode main - runOpMode - end - no InterruptedException
    11-01 12:56:19.971  5618  5618 I RobotCore: BlocksOpMode - "Tf-Object-Detect" - main/main - run2 - before clearScript
    11-01 12:56:19.971  5618  5699 V RobotCore: thread: ...terminating 'LinearOpMode main'
    11-01 12:56:19.971  5618  5687 D RobotCore: system telemetry: key=$System$Error$ msg="Fatal error occurred while executing the block labeled "call Tfod.initialize"."

cmacfarl, 2018-11-01:
That's correct.  Ultimately, createTFObjectDetector() resolves to this...

```
  /**
   * Return true if this device is compatible with Tensor Flow Object Detection, false otherwise.
   */
  public static boolean isDeviceCompatible() {
    // Requires Android 6.0+
    return Build.VERSION.SDK_INT >= Build.VERSION_CODES.M;
  }
```

Marshmallow and above.  I'll leave this issue open as documentation for anyone that stumbles across this.  However it's probably useful to update the comments in the sample to reflect this requirement.

NoahAndrews, 2018-11-01:
We should also fail with a clear error message

closed 2020-08-26 by cmacfarl