ftctechnh/ftc_app #625: TensorFlow Error "Didn't find class "com.google.ftcresearch.tfod.util.Size""

EddieDL, 2018-11-02:
When you attempt to run the TensorFlow example I get the following in the log and the OpMode shuts down.  This is on the Moto E4.

08-06 10:35:44.877  7587  7842 I AR      : ObjectTracker: This device doesn't support model targets extended tracking, extended tracking will be turned off
08-06 10:35:45.564  7587  7842 V RobotCore: thread: ...terminating 'LinearOpMode main'
08-06 10:35:45.565  7587  7842 E ThreadPool: exception thrown in thread pool; ignored
08-06 10:35:45.570  7587  7842 E ThreadPool: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/ftcresearch/tfod/util/Size;
08-06 10:35:45.573  7587  7842 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.tfod.CameraInformation.<init>(CameraInformation.java:38)
08-06 10:35:45.577  7587  7842 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.<init>(VuforiaFrameGenerator.java:63)
08-06 10:35:45.581  7587  7842 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.<init>(TFObjectDetectorImpl.java:121)
08-06 10:35:45.584  7587  7842 E ThreadPool: 	at org.firstinspires.ftc.robotcore.internal.system.ClassFactoryImpl.createTFObjectDetector(ClassFactoryImpl.java:87)
08-06 10:35:45.588  7587  7842 E ThreadPool: 	at org.firstinspires.ftc.team5391.ConceptTensorFlowObjectDetection.initTfod(ConceptTensorFlowObjectDetection.java:185)
08-06 10:35:45.592  7587  7842 E ThreadPool: 	at org.firstinspires.ftc.team5391.ConceptTensorFlowObjectDetection.runOpMode(ConceptTensorFlowObjectDetection.java:99)
08-06 10:35:45.594  7587  7842 E ThreadPool: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
08-06 10:35:45.595  7587  7842 E ThreadPool: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
08-06 10:35:45.596  7587  7842 E ThreadPool: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300)
08-06 10:35:45.597  7587  7842 E ThreadPool: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
08-06 10:35:45.598  7587  7842 E ThreadPool: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
08-06 10:35:45.599  7587  7842 E ThreadPool: 	at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
08-06 10:35:45.599  7587  7842 E ThreadPool: 	at java.lang.Thread.run(Thread.java:761)
08-06 10:35:45.619  7587  7842 E ThreadPool: Caused by: java.lang.ClassNotFoundException: Didn't find class "com.google.ftcresearch.tfod.util.Size" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk", zip file

cmacfarl, 2018-11-02:
Fixed in 4.3

https://github.com/ftctechnh/ftc_app/releases

closed 2018-11-02 by cmacfarl

EddieDL, 2018-11-03:
So I figured out my problem @cmacfarl and it isn't as simple as just upgrading. We have two teams so we do not use the standard TeamCode.

Because the change modified the build.release.gradle in TeamCode you have to make the same changes to all of your duplicate TeamCode files.  I should have been watching for it, but didn't.

I am changing the title to reflect what my real issue was.

EddieDL, 2018-11-03:
The following needs to be added to your build.release.gradle file.
    
    implementation (name: 'tfod-release', ext:'aar')
    implementation (name: 'tensorflow-lite-0.0.0-nightly', ext:'aar')