ftctechnh/ftc_app #638: Configuration not detecting external webcam Serial

webcamqn, 2018-11-12:
We are trying to use an external camera for mineral detection. The configuration file detects the webcam, however, most of the time, the identified serial number is 0.0 which causes the program to error. We are able to resolve the issue by scanning for usb devices repeatedly, but haven't been able to identify a root cause. Any suggestions would be a great help.

kev626, 2018-12-27:
My team is having this exact problem as well
**Phone**: Moto E4

**Camera** Logitech C170

**USB Hub**: Recommended Anker hub with backpowering setup

Configuration is identical

gearsincorg, 2018-12-29:
@kev626 mentioned "back powering".  
Exactly how are you both feeding power into the hub?  Are you sending power into the charger port? from where?

kev626, 2018-12-29:
I'm taking power from the +5V port on the rev hub and sending it to the charge port on the USB hub according to this guide: https://github.com/ftctechnh/ftc_app/wiki/Using-an-External-UVC-Camera-and-a-Powered-USB-Hub#usb-hub

ftctechnh, 2020-07-20:
Fixed with a pull request that has been tested and merged into the private ftc_sdk repository.

closed 2020-07-20 by ftctechnh