ftctechnh/ftc_app #639: Drive Station phone unable to find OpModes

CodingKoopa, 2018-11-13:
Our team has setup OnBot Java, and have written several OpModes for it. We were able to compile the Java code, and the OpModes were recognized as expected. However, we have been transitioning over to Android Studio, and can no longer get any options to appear in either of the two OpMode menus of the Drive Station phone. I have made sure that Instant Run is disabled in our Android Studio settings, and that the couple of testing OpModes we’ve wrote within Android Studio are marked (one with `@Autonomous`, and one with `@TeleOp`). It’s the same as the code that used to work in OnBot Java. When building and running the code, thought, nothing shows up.

Additionally, even when going back to the original OnBot Java setup, we still can’t get any OpModes to show up.

I’ll add that while setting up Android Studio, I added, compiled, and installed a class in the `org.firstinspires.ftc.teamcode.autonomous` package for organization, and without being properly marked. Both of those things have been fixed now, but I had to change the class name as there was an error about the name already being registered.

gearsincorg, 2018-11-14:
Did you remember to comment out the @disable line?

dmssargent, 2018-11-14:
What steps have you taken to revert to the OnBotJava setup?

CodingKoopa, 2018-11-27:
The issue has been resolved on the end of OnBotJava by installing the Drive Station app update, and ensuring that the phones are paired correctly. The issue was that the phone was not correctly connecting to the Robot Controller, and was silently failing to fetch the OpMode list. What exactly caused this, I’m not sure.

I’ll report back with the updated results for Android Studio as soon as I can.

CodingKoopa, 2018-12-05:
Having taken the steps previously mentioned, the Android Studio OpModes are fixed.

closed 2018-12-05 by CodingKoopa