ftctechnh/ftc_app #672: Robot Controller Error: "BasicOpMode_Iterative" class doesn't inherit form the class "OpMode

bobobachan, 2018-12-31:
After trying for 5 plus times, the robot controller still shows this error message, what should be changed in the programme? 

Platform used: Android Studio
Language used: Java
Phone used: Samsung Galaxy s5 

Thanks!

cmacfarl, 2018-12-31:
Can you please provide the full build log?

gearsincorg, 2019-01-01:
It sounds like you have a problem with the path/directory structure of your project.

When you download the SDK code, did you "import" the project into Android Studio rather than just opening it?

Did you try to build and download the app *before* making any changes?  If so, did it give the error before you made any changes?  

What if any changes have you made?