ftctechnh/ftc_app #677: Pressing the stop button starts the program

chexbox, 2019-01-05:
After initializing the program, one member of my team decided to cancel it. When he pressed stop, the program started. It seems to act this way consistently.

David10238, 2019-01-06:
We don't do the if(opModeIsActive). We just have all loops include && opModeIsActive. We may have the bot twitch, but other than that nothing.

chexbox, 2019-01-07:
Interesting solution, but it is probably not a good solution from a UX point of view. I'll have to try it anyway.

gearsincorg, 2019-01-07:
Another way to do it is to put a test just after the waitForStart() that returns immediately if stop has been pressed instead of play

waitForStart();
if (isStopRequested())
  return;

I personally don't like this approach because back in the day, when I was learning to program, it was frowned on to have more than one exit point from a function or method.

I also prefer the big if() because it ensure no further movement, and it provides the opportunity to do any shutdown after the if() to clean up anything that was started (like vision processing) before the waitForStart().

It's just a sign of more thoughtful programming to prevent your auto code running when the user decides to abort before hitting play.    

If you just let the program run, and hope for the while loops to terminate quickly, it won't stop other things like servo movement which might unlatch an arm or winch, which then may need to be reset.