ftctechnh/ftc_app #678: Using the Rev IMU to turn in Autonomous

simstrdevelopment, 2019-01-07:
Our team members are trying to code our autonomous using the built-in imu on the rev hub to turn at a certain degree or angle. With the imu sample code provided, we haven't been able to achieve

vwestin, 2019-01-07:
Once you initialize the IMU hardware/code (in init as documented in the example), you can make simple calls to get the orientation update whenever you want it:
private void checkOrientation() {
        // read the orientation of the robot
        angles = this.imu.getAngularOrientation(AxesReference.INTRINSIC, AxesOrder.ZYX, AngleUnit.DEGREES);

         this.imu.getPosition();
        // and save the heading
        curHeading = angles.firstAngle;
    }

This should work well in any autonomous code. Note that the orientation starts a 0 when the OpMode starts, so be in your starting position before you start the OpMode.

LibertyPrime15, 2019-10-01:
> Once you initialize the IMU hardware/code (in init as documented in the example), you can make simple calls to get the orientation update whenever you want it:
>
> private void checkOrientation() {
> // [...]

So that code actually works quite well and I really appreciate it, but how do you reset the imu heading once you've turned?

gunvir103, 2019-12-19:
sleep and int again is my best guess or just init

gearsincorg, 2019-12-19:
Do NOT init the gyro again.  This process does not happen immediately, so you don't want to have to deal with the init process again.

Most people just let the gyro run, and write your code so you can turn from any angle to any angle.  This way there is no need to set it back to zero each turn.

If you have a compelling reason to reset the heading to zero, simply have an intermediary variable that you use as a gyro offset.

double live_gyro_value = 0;
double gyro_offset = 0;

double getHeading( ) {
   //  read the gyro
  live_gyro_value = .... code to read gyro here.
  return (live_gyro_value - gyro_offset);
}

Then when you want to reset heading back to zero, just update the offset..

void reset_gyro( ) {
  gyro_offsett = live_gyro_value;
}

FROGbots-4634, 2019-12-19:
Another benefit to what @gearsincorg mentioned is that it's more accurate since you turn to absolute headings instead of relative which will build error with each turn.

typemismatch, 2020-01-12:
> Take a look at my library for more examples of autonomous functions. You can install the module if you want and just use the pre-built functions.
> This is the library: https://github.com/archishou/MasqLib
> Use this file to look at the functions: https://github.com/archishou/MasqLib/blob/master/src/main/java/Library4997/MasqRobot.java

This repo appears to be private?

argentox, 2020-09-07:
Thank you for your great advice. 
We want to reposition the REV Control Hub and mount it vertically. Does the IMU work equally regardless of hub orientation?

gearsincorg, 2020-09-07:
Different folks have different experiences here for some reason.  

My team routinely mounts their Expansion hubs in the vertical orientation with no problems.  Before you fully commit, I would strongly recommend doing a temporary mount, and then using one of the  sample GYRO programs to verify that the xyz axis rotations are what you expect.  Double check that when you turn the robot, you know which axis orientation is changing, and which direction gives a +ve change.

In fact, ANY time you use a sensor, it's always a good idea to do a temporary setup and look at how the data changes for your specific situation.  Understanding how a sensor responds to the real world is probably THE most important lesson that can be learned as you start prototyping your robot.