ftctechnh/ftc_app #681: UVC driver usually crashes Linux kernel if run multuiple times in a row

FROGbots-4634, 2019-01-12:
**[Here's a folder with videos of all my tests](https://drive.google.com/open?id=1QOJWZh013B30ZZEVTBrwNhvzhMdOzoa5)**

Title basically says it all.

Here's the syslog of the event:

```
[  967.760390] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  967.760452] pgd = e8e38000
[  967.760548] [00000000] *pgd=00000000
[  967.760656] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[  967.760711] CPU: 0    Not tainted  (3.4.0-g853158b #1)
[  967.760810] PC is at xhci_free_segments_for_ring+0x2c/0x88
[  967.760866] LR is at xhci_free_segments_for_ring+0x54/0x88
[  967.760964] pc : [<c0623f24>]    lr : [<c0623f4c>]    psr: a0000093
[  967.760967] sp : e83e7c18  ip : 60000093  fp : e83e7c3c
[  967.761109] r10: 00000000  r9 : 00000001  r8 : 00000000
[  967.761162] r7 : e9942000  r6 : e845f940  r5 : 00000000  r4 : 00000000
[  967.761255] r3 : 00000001  r2 : c1131c8c  r1 : 012bf000  r0 : ee002180
[  967.761353] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  967.761409] Control: 10c5787d  Table: 3123806a  DAC: 00000015
```
It traces back to:

```
#18 pc 0001da97  /data/app/com.qualcomm.ftcrobotcontroller-2/lib/arm/libRobotCore.so (_Z22uvc_user_callback_mainP17uvc_stream_handle+274)
```

Also, I did a hotplug recovery test  (this time using ZTEs since I figured the Tech Team would have verified it worked with those) and it FAILED all 3 tests I did.

cmacfarl, 2019-01-12:
Can you please clarify what you mean by "run multiple times in a row"?  -- Thanks.

FROGbots-4634, 2019-01-12:
@cmacfarl sure. So basically if I run the sample Vuforia webcam OpMode, stop it, and repeat 3 more times, the Linux kernel dies and the entire Android OS crashes, [as you can see in this video](https://drive.google.com/file/d/18lnfWtzEZq3MJRXhGRgAuJfHYmXBYoHx/view?usp=sharing).

cmacfarl, 2019-01-12:
Ack.  Thanks.

cmacfarl, 2019-01-14:
This was tested over the weekend using two Moto E4's, an Anker hub, USB battery pack, and a Logitech C920, and was not reproducible by following the procedures in the video.

It's unclear from your text above whether or not your test with ZTE's followed the procedures documented in your video or whether your hotplug description refers to something else.  Please clarify and supply an inventory of the hardware devices under test.  Thanks.

FROGbots-4634, 2019-01-15:
@cmacfarl 

Hardware used:

 - Nexus 5 as RC (tested with both official 6.0.1 and unofficial 7.1.2)
 - Logitech C920 as webcam
 - [DROK USB meter](https://www.amazon.com/DROK-Multimeter-Multifunctional-Electrical-Capacity/dp/B07MK8HGPP/ref=sr_1_7?ie=UTF8&qid=1547577734&sr=8-7&keywords=usb+analyzer)
 - [StarTech powered USB 2.0 hub](https://www.amazon.com/StarTech-com-Port-Compact-Black-USB/dp/B000T9S4CI/ref=sr_1_6?ie=UTF8&qid=1547577760&sr=8-6&keywords=startech+powered+usb+2+hub)

However, as you can see in the other videos, the crash occurs even without the meter and USB hub.

Also, to confirm that it is an issue with your UVC driver, I compiled and ran the the sample UVCCamera app which uses `libuvc` (the same library you use) and it ran multiple times with no issues at all.

Regarding the ZTE hotplug test, that is totally separate and unrelated, I was just trying to point out that there seems to be a general instability in your UVC driver.

rgatkinson, 2019-01-15:
Nit (just for posterity): the libuvc used in the FTC SDK is not bug-for-bug compatible with saki's. :-)

qwertychouskie, 2019-01-16:
This should be reported to Google as a Denial of Service (DoS) attack, a user-space app should NEVER be able to crash the kernel.  BTW, sometimes getting reboots on our S5, may be this issue.  Will test soon.

qwertychouskie, 2019-01-19:
Confirmed to reproduce on the Galaxy S5 with out team's Auto program (https://github.com/FTCTeam10298/code-2018-19).  The first 4th run we got an emergency stop, something about calling getCameraFocus (or something similar) on a null object.  We did Restart Robot, then it kept saying that the camera could not be found (as in when it is not plugged in).  Unplugging the MicroUSB OTG cord from the phone and plugging it back in cleared this.  The second 4th run, the phone froze and rebooted.  Once someone figures out what kernel code is crashing, this should be reported as a kernel bug.

float23, 2019-01-21:
we are having a similar issue with a Samsung galaxy s5 when vuforia is using a usb webcam and and any modern robotics core control module is unplugged and plugged in the phone reboots is there any way of fixing this?

qwertychouskie, 2019-01-21:
@float23 Not to be "that guy", but the Rev Expansion Hubs are way better than the Modern Robotics modules, especially when it comes to handling hot reconnects, I highly recommend the Rev hub.  We use 2 Rev hubs and the Galaxy S5s and it's great.

FROGbots-4634, 2019-01-24:
@cmacfarl @rgatkinson any update on whether this will be fixed in the foreseeable future? If not then I'll probably work on integrating my own UVC driver.

sbdevelops, 2019-04-11:
My team switched to using a Logitech C920 recently, and I've been testing autonomous over the past couple days, noticing occasional crashes when starting autonomous (which immediately is supposed to start Vuforia/TFOD). From all of my research, this driver issue seems to be the cause. I've experienced the same crash symptoms as the video produced by @FROGbots-4634. I'll post syslogs next time a crash occurs. Samsung Galaxy S5 is being used as our RC phone.

FROGbots-4634, 2019-04-11:
@sbdevelops you can ensure it won't happen during competition by never running it more than 3 times in a row without unplugging/replugging the phone.

FROGbots-4634, 2019-06-24:
@rgatkinson @cmacfarl will this be addressed in v5.x?

FROGbots-4634, 2019-09-27:
@cmacfarl @rgatkinson I have confirmed this issue still exists in SDK v5.2, and running on another device: 1st gen Pixel XL running 7.1.2. While the symptoms are not exactly the same as the Nexus 5, the UVC driver can still cause kernel to crash if run multiple times. I definitely think this warrants investigation....

FROGbots-4634, 2019-10-07:
@cmacfarl @rgatkinson this issue also affects the Moto G5 Plus, again in a slightly different manner. On the G5 I was not able to get the Linux kernel to crash, but I was able to get the OpMode to crash very occasionally with "IllegalArgumentException - pointer must not be null" but more reliably I can get the camera to fail to initialize and seemingly disappear from the USB bus and then re-appear a second later (Message "Warning: problem with Webcam 1" appears for a second).

FROGbots-4634, 2020-08-26:
Fixed in v5.5

closed 2020-08-26 by FROGbots-4634