ftctechnh/ftc_app #688: Test OpMode support

cezarmathe, 2019-01-30:
It would be really convenient to have another OpMode type for tests only that have a separate section in the Driver Station(like Autonomous and TeleOp have).

```java
package org.firstinspires.ftc.teamcode;

import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;

@Test(name = "Test example")
public class TestOpMode extends LinearOpMode {
    @Override
    public void runOpMode() {
        
    }
}
```

David10238, 2019-01-30:
This actually would be. We can easily have like a dozen of testing op modes

cezarmathe, 2019-01-30:
We're already using that feature.

cezarmathe, 2019-01-30:
It wouldn't have to be a pulldown, there could be a menu item in the menu from the top right corner which will lead to a tests screen.