ftctechnh/ftc_app #691: Config file changing motor type.

David10238, 2019-02-06:
Our config file changed the motor type of one of our drive motors from Rev to Tetrix which reverses the polarity without us changing anything in the code, wiring or configuration file. On the Discord, there are several other teams who recently experienced motors switching polarity without changing anything. My theory is they were other config files switching themselves.
We are running two rev hubs with firmware 1.8.2 and Android Studio with SDK 4.3. Our robot only has one config file.

NoahAndrews, 2019-02-06:
Hmm, interesting.

Can you think of anything you might have done around when the switch happened? Was the config correct until you loaded the configuration editor?

When was the config originally made? Did you originally make it with pre-4.0 software?

David10238, 2019-02-06:
The config was made with 4.x versions.
The config as far as we know was incorrect before we loaded it. We noticed it only when the drivetrain immediately locked up from fighting motors, and we opened it using the FTC app to check. I believe this was the first time running the bot after a competition. When we changed the motor type back to Rev it immediately worked like expected. I believe the hardware team made some minor repairs, but there were no changes to the drivetrain.

NoahAndrews, 2019-02-06:
I'm not sure that it will be helpful, but can you provide a log file from the robot controller phone?

NoahAndrews, 2019-02-06:
What was the motor type? Core Hex, Core HD 40, or Core HD 60?

David10238, 2019-02-06:
HD 20 is what it was supposed to be. I'll get a log when I have access to the phone.

NoahAndrews, 2019-02-06:
Yeah sorry, I misspoke on the motor types.

NoahAndrews, 2019-02-06:
Was it specifically the Tetrix motor, or was it "Unspecified Motor"?

David10238, 2019-02-06:
specifically Tetrix

David10238, 2019-02-07:
[robotControllerLog.txt](https://github.com/ftctechnh/ftc_app/files/2839360/robotControllerLog.txt)