ftctechnh/ftc_app #693: Problem with connecting RC and DS apps over 5ghz

BraveBots041, 2019-02-11:
Hi! I'm Dragos and i've encountered a problem with the connection of the two official apps over 5ghz. We are using two Samsung S5 phones, and they are connecting over 5ghz with no problem throuch wi-fi direct. But when we start the ftc apps, the phones are disconnecting, and reconnecting again over 2.4ghz. other times, they remain connected, but in the ftc apps, we still cannot choose a chanel greater than 11 (a 2.4ghz channel). I hope you can help us. 
We use the last version of the apps, v4.3.
[image]
[image]

gearsincorg, 2019-02-11:
Are the phones disconnecting while you are sitting in idle?

The 5GHz phones seem to take a break and scan the 5GHz spectrum if there is not a lot of app based activity.  This can cause a disconnect and the phone go back to "stopped".

The easiest way to eliminate this is to send telemetry messages all the time.
Instead of doing a waitForStart() you need to do a while loop.. with a telemetry statement in it..

eg:  just from memory.....

while(!opModeIsActive()) {
  telemetry.addData("status", "Ready");
  telemetry.update();
}

BraveBots041, 2019-02-11:
The phones are disconnecting as soon as we open the apps, and reconnecting over a 2.4ghz channel. We don't gen to initialize or choose the code in the first place. The most annoying part is that we don't even see the option to choose 5ghz channels in the apps.

qwertychouskie, 2019-02-11:
AFAIR this is our team's procedure for the S5s:

1. Make sure the phones are paired in the DS app
2. Connect to a 5GHz WiFi AP on the Driver Station phone (only needs to be done once, so long as you don't connect to a 2.4GHz WiFi AP later)
3. Use the Android settings app to connect to the Robot Controller phone
4. Disconnect from the 5GHz AP from step 2 (if applicable)
5. Open the Driver Station app
6. Profit???

You may need to make sure beforehand that the DS channel selector is set to auto/unselected/etc.

My memory is a bit fuzzy on some of this, I'll post here once I can confirm the details.  Really the DS should give the option for 5GHz channels on the S5 though.  @gearsincorg / @cmacfarl Any idea why the S5s don't show 5GHz channels in the DS app?