ftctechnh/ftc_app #702: Expansion Hubs unresponsive after ESD event until opmode is stopped

FROGbots-4634, 2019-03-07:
I'm doing some ESD testing with our robot and just ran into an extremely strange issue. After driving around the field and zapping the lander, the Expansion Hubs became unresponsive to the gamepads and the robot began stalling out against the playing field wall. However, no comms errors showed up in the log, and upon pressing the stop button, the robot _immediately_ stopped. What the heck?

Here's the log in case there's anything of interest that I missed, but I don't see anything:

```03-07 11:03:24.433  2008  2565 V Robocol : received command: CMD_RUN_OP_MODE(10420) A - Practice TeleOp
03-07 11:03:24.446  2008  2602 V Robocol : sending CMD_NOTIFY_RUN_OP_MODE(570), attempt: 0
03-07 11:03:24.577  2008  2572 V RobotCore: nack rec'd mod=3 msg#=193 ref#=0 reason=MOTOR_NOT_CONFIG_BEFORE_ENABLED:50
03-07 11:03:24.579  2008  2657 E LynxMotor: exception thrown during lynx communication
03-07 11:03:24.581  2008  2657 E LynxMotor: com.qualcomm.hardware.lynx.LynxNackException: (DQ2ASZXG #3) LynxSetMotorChannelEnableCommand: nack received: MOTOR_NOT_CONFIG_BEFORE_ENABLED:50
03-07 11:03:24.588  2008  2657 E LynxMotor: 	at com.qualcomm.hardware.lynx.commands.LynxRespondable.throwIfNack(LynxRespondable.java:325)
03-07 11:03:24.589  2008  2657 E LynxMotor: 	at com.qualcomm.hardware.lynx.commands.LynxRespondable.send(LynxRespondable.java:239)
03-07 11:03:24.591  2008  2657 E LynxMotor: 	at com.qualcomm.hardware.lynx.LynxDcMotorController.internalSetMotorEnable(LynxDcMotorController.java:221)
03-07 11:03:24.592  2008  2657 E LynxMotor: 	at com.qualcomm.hardware.lynx.LynxDcMotorController.internalSetMotorPower(LynxDcMotorController.java:543)
03-07 11:03:24.595  2008  2657 E LynxMotor: 	at com.qualcomm.hardware.lynx.LynxDcMotorController.setMotorMode(LynxDcMotorController.java:373)
03-07 11:03:24.596  2008  2657 E LynxMotor: 	at com.qualcomm.robotcore.hardware.DcMotorImpl.internalSetMode(DcMotorImpl.java:322)
03-07 11:03:24.597  2008  2657 E LynxMotor: 	at com.qualcomm.robotcore.hardware.DcMotorImpl.setMode(DcMotorImpl.java:318)
03-07 11:03:24.598  2008  2657 E LynxMotor: 	at net.frogbots.roverruckus.hardware.components.DebrisAssembly.init(DebrisAssembly.java:35)
03-07 11:03:24.599  2008  2657 E LynxMotor: 	at net.frogbots.roverruckus.hardware.Robot.init(Robot.java:39)
03-07 11:03:24.601  2008  2657 E LynxMotor: 	at net.frogbots.roverruckus.meta.opmode.FrogOpModeImpl.runOpMode(FrogOpModeImpl.java:68)
03-07 11:03:24.601  2008  2657 E LynxMotor: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
03-07 11:03:24.602  2008  2657 E LynxMotor: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
03-07 11:03:24.602  2008  2657 E LynxMotor: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:300)
03-07 11:03:24.603  2008  2657 E LynxMotor: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
03-07 11:03:24.603  2008  2657 E LynxMotor: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
03-07 11:03:24.604  2008  2657 E LynxMotor: 	at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
03-07 11:03:24.604  2008  2657 E LynxMotor: 	at java.lang.Thread.run(Thread.java:761)
03-07 11:03:24.753  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:24.754  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:24.754  2008  2657 E LynxI2cDeviceSynch: placeholder: readStatusQuery
03-07 11:03:24.821  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:24.821  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:24.888  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:24.888  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:24.955  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:24.955  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:25.022  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:25.023  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:25.091  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:25.092  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:25.160  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:25.160  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:03:25.228  2008  2657 V RobotCore: addr=false data=true arb=false clock=false
03-07 11:03:25.228  2008  2657 E LynxI2cDeviceSynch: readStatusQuery: cbExpected=1 cbRead=0
03-07 11:04:00.916  2008  2565 V Robocol : received command: CMD_INIT_OP_MODE(11619) $Stop$Robot$
```

Also, in a previous test, the gamepad disconnected from the driver station at the exact moment of the ESD event on the robot. Coincidence? Somehow I think not. Unfortunately I do not have the logs for that encounter because they were overwritten by the log from another test.

Both Expansion Hubs are running firmware 1.8.2

NoahAndrews, 2019-03-07:
I literally was just looking at possibly this very thing. Are you using run to position mode?

FROGbots-4634, 2019-03-07:
I was able to catch the gamepad disconnecting from the driver station during the ESD event in the act in a log!

```
03-07 11:31:16.205  3469  3576 V Robocol : received command: CMD_INIT_OP_MODE(10803) A - Practice TeleOp
03-07 11:31:16.234  3469  3575 I RobotCore: ******************** START - OPMODE A - Practice TeleOp ********************
03-07 11:31:16.235  3469  3595 I RobotCore: Attempting to switch to op mode A - Practice TeleOp
03-07 11:31:16.314  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:16.318  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:16.321  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:16.324  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:16.328  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:16.331  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:16.334  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:16.338  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:16.544  3469  3595 D STMicroVL53L0X: : Checking to see if it's really a VL53L0X sensor...
03-07 11:31:16.554  3469  3595 D STMicroVL53L0X: : Reg 0xC0 = ee (should be 0xEE)
03-07 11:31:16.562  3469  3595 D STMicroVL53L0X: : Reg 0xC1 = aa (should be 0xAA)
03-07 11:31:16.571  3469  3595 D STMicroVL53L0X: : Reg 0xC2 = 10 (should be 0x10)
03-07 11:31:16.580  3469  3595 D STMicroVL53L0X: : Reg 0x51 = 0 (should be 0x0099)
03-07 11:31:16.590  3469  3595 D STMicroVL53L0X: : Reg 0x61 = 0 (should be 0x0000)
03-07 11:31:16.670  3469  3595 D STMicroVL53L0X: : initial sig rate lim (MCPS) 0.000000
03-07 11:31:16.685  3469  3595 D STMicroVL53L0X: : adjusted sig rate lim (MCPS) 0.250000
03-07 11:31:17.369  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:17.379  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:17.383  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:17.387  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:17.392  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:17.398  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:17.402  3469  3595 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
03-07 11:31:17.407  3469  3595 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
03-07 11:31:17.428  3469  3739 V RobotCore: thread: 'LinearOpMode main' starting...
03-07 11:31:17.430  3469  3629 V Robocol : sending CMD_NOTIFY_INIT_OP_MODE(949), attempt: 0
03-07 11:31:19.682  3469  3576 V Robocol : received command: CMD_RUN_OP_MODE(10887) A - Practice TeleOp
03-07 11:31:19.710  3469  3629 V Robocol : sending CMD_NOTIFY_RUN_OP_MODE(1020), attempt: 0
03-07 11:31:19.806  3469  3588 V RobotCore: nack rec'd mod=3 msg#=102 ref#=0 reason=MOTOR_NOT_CONFIG_BEFORE_ENABLED:50
03-07 11:31:19.806  3469  3739 E LynxMotor: exception thrown during lynx communication
03-07 11:31:19.808  3469  3739 E LynxMotor: com.qualcomm.hardware.lynx.LynxNackException: (DQ2ASZXG #3) LynxSetMotorChannelEnableCommand: nack received: MOTOR_NOT_CONFIG_BEFORE_ENABLED:50
03-07 11:31:19.810  3469  3739 E LynxMotor: 	at com.qualcomm.hardware.lynx.commands.LynxRespondable.throwIfNack(LynxRespondable.java:325)
03-07 11:31:19.810  3469  3739 E LynxMotor: 	at com.qualcomm.hardware.lynx.commands.LynxRespondable.send(LynxRespondable.java:239)
03-07 11:31:19.812  3469  3739 E LynxMotor: 	at com.qualcomm.hardware.lynx.LynxDcMotorController.internalSetMotorEnable(LynxDcMotorController.java:221)
03-07 11:31:19.813  3469  3739 E LynxMotor: 	at com.qualcomm.hardware.lynx.LynxDcMotorController.internalSetMotorPower(LynxDcMotorController.java:543)
03-07 11:31:19.814  3469  3739 E LynxMotor: 	at com.qualcomm.hardware.lynx.LynxDcMotorController.setMotorMode(LynxDcMotorController.java:373)
03-07 11:31:19.816  3469  3739 E LynxMotor: 	at com.qualcomm.robotcore.hardware.DcMotorImpl.internalSetMode(DcMotorImpl.java:322)
03-07 11:31:19.817  3469  3739 E LynxMotor: 	at com.qualcomm.robotcore.hardware.DcMotorImpl.setMode(DcMotorImpl.java:318)
03-07 11:31:19.818  3469  3739 E LynxMotor: 	at net.frogbots.roverruckus.hardware.components.DebrisAssembly.init(DebrisAssembly.java:35)
03-07 11:31:19.818  3469  3739 E LynxMotor: 	at net.frogbots.roverruckus.hardware.Robot.init(Robot.java:39)
03-07 11:31:19.818  3469  3739 E LynxMotor: 	at net.frogbots.roverruckus.meta.opmode.FrogOpModeImpl.runOpMode(FrogOpModeImpl.java:68)
03-07 11:31:19.819  3469  3739 E LynxMotor: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:305)
03-07 11:31:19.819  3469  3739 E LynxMotor: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
[... cut ...]
03-07 11:31:50.937  3469  3595 I FtcEventLoop: ^^===== MODULE ATTACH: complete REV Robotics USB Expansion Hub Portal [DQ2ASZXG]=====^^
03-07 11:33:09.723  3469  3576 V EventLoopManager: synthetic gamepad received: id=-2 user=ONE atRest=true 
03-07 11:33:14.044  3469  3576 V Robocol : received command: CMD_INIT_OP_MODE(14368) $Stop$Robot$
03-07 11:33:14.061  3469  3575 I RobotCore: ******************** STOP - OPMODE /storage/emulated/0/FIRST/matchlogs/Match-18-A_-_Practice_TeleOp.txt ********************
```

FROGbots-4634, 2019-03-07:
> I literally was just looking at possibly this very thing. Are you using run to position mode?

Yes I am, for more than one motor actually.

NoahAndrews, 2019-03-07:
Make sure you always assign a position to the motor before putting it in run to position mode. Do you still see the issue when that happens?

FROGbots-4634, 2019-03-07:
> Make sure you always assign a position to the motor before putting it in run to position mode. Do you still see the issue when fixing that?

This issue is pretty hard to reproduce, it might take me a while to determine the answer to that...

FROGbots-4634, 2019-03-07:
Also, here's the DS's log of the gamepad disconnect during the ESD event. (Not sure if it shows anything the RC log doesn't)

```
03-07 11:31:15.109 17059 17175 V Robocol : { 4606 19.714} received command: CMD_NOTIFY_RUN_OP_MODE(1020) A - Practice TeleOp
03-07 11:31:15.109 17059 17174 V DriverStation: { 4606 19.715} Robot Controller starting op mode: A - Practice TeleOp
03-07 11:31:15.110 17059 17174 V DriverStation: { 4606 19.715} ui:uiWaitingForStopEvent
03-07 11:31:19.114 17059 17087 V NetDiscover_wifiDirectAgent_bcast: { 4604 23.718} peers found: count=1
03-07 11:31:19.114 17059 17087 V NetDiscover_wifiDirectAgent_bcast: { 4604 23.718}     peer: 8e:3a:e3:71:77:7c 4634-RC
03-07 11:31:19.130 17059 17059 D WifiDirect: { 4605 23.735} broadcast: peers changed
03-07 11:31:19.147 17059 17059 V WifiDirect: { 4605 23.752} peers found: 1
03-07 11:31:19.147 17059 17059 V WifiDirect: { 4605 23.752}     peer: 8e:3a:e3:71:77:7c 4634-RC
03-07 11:31:19.148 17059 17059 I RobotCore: { 4605 23.752} Received networkConnectionEvent: PEERS_AVAILABLE
03-07 11:31:45.302 17059 17175 E DriverStation: { 4605 49.907} System telemetry warning: REV Robotics USB Expansion Hub Portal [DQ2ASZXG] detached
03-07 11:31:45.467 17059 17175 V Robocol : { 4604 50.071} received command: CMD_PLAY_SOUND(2009) {"hashString":"1e48ddeaa00488b1e532f1de418fdb6f","loopControl":0,"msPresentationTime":1551976310045,"rate":1.0,"volume":1.0,"waitForNonLoopingSoundsToFinish":true}
03-07 11:31:45.468 17059 17174 W AudioTrack: AUDIO_OUTPUT_FLAG_FAST denied by client; transfer 4, track 44100 Hz, output 48000 Hz
03-07 11:31:45.473 17059 17174 D SoundPlayer: { 4604 50.077} playing volume=1.000000 samp=8|ms=1044
03-07 11:33:05.080 17059 17059 V DriverStation: { 4605  9.685} Input device (id = 21) removed.
03-07 11:33:05.116 17059 17180 V GamepadManager: { 4605  9.721} transmitting synthetic gamepad user=ONE
03-07 11:33:05.398 17059 17059 V DriverStation: { 4609 10.007} New input device (id = 22) detected.
03-07 11:33:09.367 17059 17059 V StateMachine: { 4610 13.977} State with no transitions: WifiState
03-07 11:33:09.410 17059 17059 V StateMachine: { 4609 14.019} State with no transitions: WifiState
03-07 11:33:09.436 17059 17180 V Robocol : { 4609 14.045} sending CMD_INIT_OP_MODE(14368), attempt: 0
03-07 11:33:09.631 17059 17175 V Robocol : { 4605 14.236} received command: CMD_NOTIFY_INIT_OP_MODE(4618) $Stop$Robot$
03-07 11:33:09.632 17059 17174 V DriverStation: { 4605 14.237} Robot Controller initializing op mode: $Stop$Robot$
03-07 11:33:09.632 17059 17174 V DriverStation: { 4605 14.237} ui:uiWaitingForInitEvent
03-07 11:33:09.633 17059 17174 V StateMachine: { 4605 14.238} State with no transitions: WifiState
03-07 11:33:09.636 17059 17180 V Robocol : { 4605 14.241} sending CMD_RUN_OP_MODE(14374), attempt: 0
03-07 11:33:09.667 17059 17175 V Robocol : { 4606 14.273} received command: CMD_NOTIFY_RUN_OP_MODE(4624) $Stop$Robot$
03-07 11:33:09.668 17059 17174 V DriverStation: { 4606 14.274} Robot Controller starting op mode: $Stop$Robot$
```

NoahAndrews, 2019-03-07:
It doesn't sound like the `MOTOR_NOT_CONFIG_BEFORE_ENABLED` nacks are relevant to this bug, so nevermind what I said earlier.