ftctechnh/ftc_app #704: App crashed due to android.jar permission issue from OnBotJavaManager

twang11, 2019-03-20:
This happened on some of my E4 phones and all of my G4 phones. App crashed right after pushing the new robot controller app.

Phones are running Android 7.1.1 (version 25)

```
     Caused by: java.io.FileNotFoundException: /storage/emulated/0/FIRST/java/lib/android.jar (Permission denied)
        at java.io.FileOutputStream.open(Native Method)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:221)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:169)
        at org.firstinspires.ftc.robotcore.internal.system.AppUtil.copyStream(AppUtil.java:412)
        at org.firstinspires.ftc.robotcore.internal.opmode.OnBotJavaManager.extractJavaLibraryAsset(OnBotJavaManager.java:239)
```

NoahAndrews, 2019-03-20:
It sounds like you increased the SDK target version in response to an Android Studio warning. Set the SDK target back to 19. Future releases will target 26.

twang11, 2019-03-20:
reversed back to 19, worked. Wondering why target 26 worked for some E4 phones (2 out of 4).

closed 2019-03-20 by twang11

cmacfarl, 2019-03-20:
@twang11 The answer is wrapped up in Google's permissions model.  

More information than you probably really want here.

https://developer.android.com/guide/topics/permissions/overview

Next year's SDK and Driver Station, will by necessity, e.g. Google decree, ask for permissions when first run.  In the meantime don't push the target sdk past 23 and run on an Android 6.0+ phone.