ftctechnh/ftc_app #708: Gyros on Ford Field

David10238, 2019-04-29:
On Ford Field, our gyros had problems during calibration time. We got the "problem with imu" error. We talked with BrainStem who said they had it last year, and we fixed the issue by turning the phone on and opening the app before plugging in the bot. We believe the issue is static messing with the hubs.

gearsincorg, 2019-04-29:
My recommendation is always to have the 12V powered ON before connecting the RC phone. (this ensures that the expansion hubs are never powered by only the phone's 5V (which is not truly able to power two hubs).

Then once the 12V is on, I recommend NOT starting the app before plugging in the phone.  I like to ensure that the app has been swiped off (fully out of memory) and then plugging in the phone to the robot and watching the App auto launch.  This ensures that there is a good USB connection (power and data) .  It also ensures that the control hubs are fully ready when the app starts so there should be NO errors appearing on the screen when the app has finished loading.

ANY error or warning that appears is now REAL and must be addressed.

Note: "Problem with imu" can often occur when the the coach hits the init button several times quickly by mistake.  Recovery is always to power down the 12V, disconnect the phone, swipe off the app and startup as described above.

David10238, 2019-04-29:
On Ford Field the process they had us do certain things at certain times to make things go smoothly.
First, turn phones on, start apps, change Wifi-Channel, select an OpMode and leave in that state.
Second, put everything on the field with the power off
Then about a minute before the match turn on battery, initiate the program and hang bot.

FROGbots-4634, 2019-04-29:
> My recommendation is always to have the 12V powered ON before connecting the RC phone

**This --^** (also make sure to unplug phone before turning off 12v)

If you have a lot of sensors/encoders plugged in, not only will the phone not be able to power everything, but also it will be unable to detect any USB devices until it is cold rebooted. Of course, YMMV.

----

> Note: "Problem with imu" can often occur when the the coach hits the init button several times quickly by mistake

I second this also, and it can cause problems with more than just the IMU.

ftctechnh, 2019-05-02:
Hi David10238,

I'm sorry that you had this issue with the IMU on Ford field.  Thanks for posting information about this issue here.

I have seen examples where the "problem with IMU" error seemed to be caused by something disrupting the I2C port bus #0 on the REV Expansion Hub.  In bench top testing and also at events,  I have observed an electrostatic discharge event disrupt the operation of the I2C port.  however, as gearsincorg mentioned, there are other scenarios that could also cause the error.

For your robot that had the issues on Ford field, did the robot have a REV Robotics grounding strap installed (REV-31-1269)?  Also, did that I2C bus #0 port have any devices installed?  When you received the error while calibrating your sensors, was there a specific event (like hanging the robot on the lander, or having a team member touch the metal frame of the robot) that occurred that suggested an electrostatic discharge happened?

I have seen instances this past season where teams got the "problem with imu" error message when, for example, they went to hang their robot on the lander before a match.  Installing a grounding strap or insulating the metal components on the robot (with electrical tape) addressed the issue for the instances that I witnessed.  Note that I saw these errors  at events where the tiles were **not** treated with the recommended anti-static spray.

Note that for Houston and Detroit this year, strong steps were taken to mitigate ESD effects on the competition fields (in the convention halls and at the stadiums).  A recommended heavy duty commercial anti-static spray was liberally applied to the fields to reduce the likelihood of charge build up on the robots.  These steps, however, might not completely eliminate all of the charge build up on or around the field.

Congrats on making it to Ford Field.  I hope you had a fun experience in Detroit!

Tom

David10238, 2019-05-02:
We did have a grounding strap installed. As far as I can tell the only difference between when it worked and when it didn't work was the order we did things in. If we had the app opened before plugging the phone into the bot it worked, if not we got the "problem with imu". A lot of our wires run through wire loom, and the hubs are mounted straight to Lexan. This season we never had any ESD issues causing an actual dc even in practice. I don't think we had any extra I2C devices connected to the bus, but we may have had a Rev TOF connected. I know Gluten Free had ESD during finals 3 as well, and that caused their hang encoder to be slightly off. There are also several places we could have ESD cause issues, the field never shocked me, but the pits and bleacher areas were loaded with static shocks. Our initialization process just has us placing the bot on the field, adjusting the angle and pressing initialization, we then watch to make sure no errors pop up before hanging(we never even on Ford Field saw "problem with imu" after this step).
I am from 7236

ftctechnh, 2019-05-06:
Thanks for the detailed info David.  Would it be possible to get a copy of your robot controller's log files and the approximate time in the log when the disconnect occurred?  

Could you ZIP the files (the system actually saves the robotControllerLog.txt file as well as 4 cyclic log files in /sdcard on the Android device) and post them here or on a shared drive?

Thanks and Congratulations again!

Tom