ftctechnh/ftc_app #713: Incorrect motor configuration data

David10238, 2019-06-06:
The ticks per rev of RevRobotics40HDHexMotor and RevRobotics20HDHexMotor are twice what they should be
RevRobotics40HDHexMotor should be 1120 not 2240
RevRobotics40HDHexMotor should be 560 not 1120

FROGbots-4634, 2019-06-06:
Well, assuming that the encoder on the REV HD Hex motor has a 7-pole magnet, it appears you are correct.

----

Calculations with 40:1 reduction

```
ChanA cycles per rev at encoder shaft = 7

 7 *  2 = 14    (cycles of A and B)
14 *  2 = 28    (edges of a pulse)
28 * 40 = 1120  (edges * gearbox reduction) --> TOTAL = 1120
```

Calculations with 20:1 reduction

```
ChanA cycles per rev at encoder shaft = 7

 7 *  2 = 14   (cycles of A and B)
14 *  2 = 28   (edges of a pulse)
28 * 20 = 560  (edges * gearbox reduction) --> TOTAL = 560
```

NoahAndrews, 2019-06-07:
The REV [encoder guide](http://www.revrobotics.com/content/docs/Encoder-Guide.pdf) and a quick test confirms that you're correct. I'll submit a PR to get this fixed.

closed 2019-06-07 by cmacfarl

rbrott, 2019-06-15:
I know this is probably fixed in the internal repo, but it would be nice to keep the issue open here until it's actually fixed in a release (especially since releases are now few and far between).

cmacfarl, 2019-06-15:
I can confirm that it's fixed in v5.0 which will be released within the
next couple weeks.

rbrott, 2019-06-15:
I understand, but it's still present in the current version. The open issues on this repo are the canonical (and pretty much sole) list of known issues with the SDK. The open public issues should reflect the state of the public SDK not the internal SDK.

rbrott, 2019-06-15:
Additionally, it would be nice to have a list of fixed public GitHub issues in the release changelogs (or some kind of milestone/label). It could also help clarify situations like [this](https://github.com/ftctechnh/ftc_app/issues/614).