ftctechnh/ftc_app #714: How to send data through i2c? trying to communicate with Pixy2 camera

ftc12265, 2019-06-23:


NoahW314, 2019-06-24:
[This FTC forum post](https://ftcforum.usfirst.org/forum/ftc-technology/android-studio/7211-i2c) has some information and code that should help you communicate with the Pixy2.

FROGbots-4634, 2019-06-24:
You cannot use the Pixy2 in the default I2C mode. You need to use it in Lego mode. (Default I2C mode doesn't work because of the extremely high (when compared to an Arduino) latency of I2C operations through the SDK.

FROGbots-4634, 2019-06-24:
Also keep in mind that the user-created firmware created for FTC that allows the Lego mode to return more data is [illegal for competition](https://ftcforum.usfirst.org/forum/tech-challenge-rover-ruckus-presented-by-qualcomm/robot-inspection-and-build-rules-aa/robot-electrical-parts-and-materials/65389-sensors-answers?p=66223#post66223)