ftctechnh/ftc_app #715: OnBot SDK is different than Android Studio SDK

stemrobotics, 2019-06-24:
I have example for code for students that compiles without problem in Android studio but does not compile in OnBot due to differences in the SDK libraries. For instance, the items hardware.Servo.MAX_POSITION & MIN_POSITION is not found on OnBot but resolves correctly in AS. I  have found several of these cases. What is happening here and how would I correct this situation so that my code set works in both environments?

NoahAndrews, 2019-06-24:
This is a known issue that is being tracked internally. It's good to know that someone has run into it in the real world.

stemrobotics, 2019-06-25:
So what does tracked internally mean? This seems a siginificant issue to me and since I am trying to use the same code both ways, I need to understand what your plans are to fix this. I am updating a curriculum we use for FIRST robotics programs in our school district...

Richard Corn 
Software Devices LLC 
P.O. Box 12299 
Olympia, WA. 98508   360.357.9572 
rac@swdevs.com     www.swdevs.com    
  ----- Original Message ----- 
  From: Noah Andrews 
  To: ftctechnh/ftc_app 
  Cc: StemRobotics ; Author 
  Sent: Monday, June 24, 2019 12:33 PM
  Subject: Re: [ftctechnh/ftc_app] OnBot SDK is different than Android Studio SDK (#715)

  This is a known issue that is being tracked internally. It's good to know that someone has run into it in the real world.

FROGbots-4634, 2019-06-25:
> So what does tracked internally mean?

It means they have an issue open about it in their private development repository.

gearsincorg, 2019-06-26:
One reason that you sometimes see differences between OBJ and AS is that some libs do not fully update when you update the App version. 

This problem is corrected by completely resetting the OnBotJava environment, which clears out any old libs.

Note: Before you do this, make sure you have backed up / downloaded any opmodes that are on your phone, that you want to keep.  All code will be deleted from the phone during the reset.  Just select them on the file tree, do a right click, and select download.

Easy steps:
Connect your PC to the phone, and bring up the OnBotJava page.
Find the Settings (GEARS) icon at the upper left of the page.  Click it.
Scroll down and expand the "Advanced Settings" section
Click on the Red RESET button (after backing up your code)
   Since this step clears out all your programs, you are prompted TWICE if you really want to do this.

Once it's done, I'd close out the RC app (swipe it off) and restart it.
Now reconnect your PC, Upload you java files, and try rebuilding.

stemrobotics, 2019-06-26:
This answer seems to make sense and I was hoping there was just such a solution, however it did not work. 

Based on the explanation that there is a pre-compiled form of the SDK as part of the controller components, clearing things out and restarting the same controller apk or downloading a new apk via AS compile (from the existing SDK), the same pre-compiled components would be downloaded. It seems there needs to be a fix to the pre-compiled SDK component in order to resolve this problem. It appears the pre-compiled components are not rebuilt when you compile in AS, but are built when the SDK is prepared for release and do not change after that.

Richard Corn 
Software Devices LLC 
P.O. Box 12299 
Olympia, WA. 98508   360.357.9572 
rac@swdevs.com     www.swdevs.com    
  ----- Original Message ----- 
  From: Phil Malone 
  To: ftctechnh/ftc_app 
  Cc: StemRobotics ; Author 
  Sent: Wednesday, June 26, 2019 6:39 AM
  Subject: Re: [ftctechnh/ftc_app] OnBot SDK is different than Android Studio SDK (#715)

  One reason that you sometimes see differences between OBJ and AS is that some libs do not fully update when you update the App version.

  This problem is corrected by completely resetting the OnBotJava environment, which clears out any old libs.

  Note: Before you do this, make sure you have backed up / downloaded any opmodes that are on your phone, that you want to keep. All code will be deleted from the phone during the reset. Just select them on the file tree, do a right click, and select download.

  Easy steps:
  Connect your PC to the phone, and bring up the OnBotJava page.
  Find the Settings (GEARS) icon at the upper left of the page. Click it.
  Scroll down and expand the "Advanced Settings" section
  Click on the Red RESET button (after backing up your code)
  Since this step clears out all your programs, you are prompted TWICE if you really want to do this.

  Once it's done, I'd close out the RC app (swipe it off) and restart it.
  Now reconnect your PC, Upload you java files, and try rebuilding.

David10238, 2019-06-26:
I experienced this issue in January at our league championship. Trying to write a teams auto I had issues with tensorflow not working on OnBot.

FROGbots-4634, 2019-06-26:
> Based on the explanation that there is a pre-compiled form of the SDK as part of the controller components, clearing things out and restarting the same controller apk or downloading a new apk via AS compile (from the existing SDK), the same pre-compiled components would be downloaded.

Well, not exactly, because the precompiled part used by OnBot (`onbotjava-classes.jar`) is shipped as a "resource" file in the APK and then extracted to `/sdcard/FIRST` for use by `javac `later. So if the new version of the SDK sees that the JAR already exists and doesn't overwrite it, then you have a mismatched version.

> It seems there needs to be a fix to the pre-compiled SDK component in order to resolve this problem.

Based on Noah's response that this is a bug tracked in the private development repository, that appears to be correct

> It appears the pre-compiled components are not rebuilt when you compile in AS, but are built when the SDK is prepared for release and do not change after that.

Correct

stemrobotics, 2019-06-26:
Well, not exactly, because the precompiled part used by OnBot (onbotjava-classes.jar) is shipped as a "resource" file in the APK and then extracted to /sdcard/FIRST for use by javac later. So if the new version of the SDK sees that the JAR already exists and doesn't overwrite it, then you have a mismatched version.

    So I deleted the onbotjava-classes.jar and then compiled the controller project in AS and downloaded the apk. onbotjava-            classes.jar reappeared, so the jar on the phone must be the same as the jar embedded in the compiled controller package.         The compile in onbot still fails, so that would seem to confirm there is a mismatch between the SDK as used by AS to compile     and the SDK components that are pre-compiled into the resource file.

So I think the ball is in the SDK developers court.

Richard Corn 
Software Devices LLC 
P.O. Box 12299 
Olympia, WA. 98508   360.357.9572 
rac@swdevs.com     www.swdevs.com    
  ----- Original Message ----- 
  From: FROGbots-4634 
  To: ftctechnh/ftc_app 
  Cc: StemRobotics ; Author 
  Sent: Wednesday, June 26, 2019 1:32 PM
  Subject: Re: [ftctechnh/ftc_app] OnBot SDK is different than Android Studio SDK (#715)

    Based on the explanation that there is a pre-compiled form of the SDK as part of the controller components, clearing things out and restarting the same controller apk or downloading a new apk via AS compile (from the existing SDK), the same pre-compiled components would be downloaded.

  Well, not exactly, because the precompiled part used by OnBot (onbotjava-classes.jar) is shipped as a "resource" file in the APK and then extracted to /sdcard/FIRST for use by javac later. So if the new version of the SDK sees that the JAR already exists and doesn't overwrite it, then you have a mismatched version.

    It seems there needs to be a fix to the pre-compiled SDK component in order to resolve this problem.

  Based on Noah's response that this is a bug tracked in the private development repository, that appears to be correct

    It appears the pre-compiled components are not rebuilt when you compile in AS, but are built when the SDK is prepared for release and do not change after that.

  Correct

cmacfarl, 2019-06-26:
Thank you Richard, there's a problem with the generation of the the stubjars jar wherein public static fields were getting inlined by the javac compiler.   A fix, to generate code that will prevent the inlining is in progress and should appear in v5.1 which is currently targeted for late July.

In the meantime, the stubjars software is a utility located here: https://github.com/dmssargent/StubJars  That software generates the jar file that is subsequently packaged with the released SDK.

Note the develop branch has the commits that in process for inclusion in the SDK.

dmssargent, 2019-06-27:
If you need an immediate workaround, reflectively accessing the fields in question will still function just fine.

And although I will admit it really may not be obvious (and it will be gross), the StubJars library and the ftc_app contain enough information, such that you can fix the error until we release v5.1.

I might write a script to automate the regeneration and correct update of the OnBotJava SDK if the need is great.

stemrobotics, 2019-06-27:
Thanks for replying with recongition of the problem, some details as to the cause of the issue and a fix. I'm happy to wait for v5.1 release.

Richard  Corn

  SDK (#715) 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ftctechnh/ftc_app/issues/715?email_source=notifications\u0026email_token=AHSODHUA7V4QSBZU7H7YB2LP4QDOXA5CNFSM4H3BOVAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYVF62A#issuecomment-506093416",
"url": "https://github.com/ftctechnh/ftc_app/issues/715?email_source=notifications\u0026email_token=AHSODHUA7V4QSBZU7H7YB2LP4QDOXA5CNFSM4H3BOVAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYVF62A#issuecomment-506093416",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

stemrobotics, 2019-06-27:
An related question for the developers: I am wondering how we go from current release of 4.3 to the coming 5.1? What happened to 5.0?

cmacfarl, 2019-06-27:
5.0 is a prerelease for the Control Hub pilot purposes.   It should be up on ftc_app in the next week or so.  There are a couple new interesting features, most notably, a Java preview pane for the Blocks editor, and an offline mode for Blocks.  But the big change is full support for the Control Hub.