ftctechnh/ftc_app #740: Robot Controller does not detect more than 1 device

serb1231, 2019-11-17:
I have connected 2 Core Motor Controllers and 1 Core Device Interface to the Core Power Distribution Module(each with motors and servo motors connected) and when I scan it, the FTC Apps only see the Core device Interface(when all 3 are connected), only 1 Core Motor Controller (the same Core Motor Controller every time) when the Device Interface isn't connected and the other Motor Controller when only it is connected.

FROGbots-4634, 2019-11-17:
What version of Android are you using? Some builds of Android 5.0 have a bug where only one USB device will be detected.

serb1231, 2019-11-17:
Hello,thank you for your response. We have the 5.3 version(the latest one)

dum., 17 nov. 2019, 4:25 p.m. FROGbots-4634 <notifications@github.com> a
scris:

NoahAndrews, 2019-11-17:
He was asking what version of **Android**, not which version of the FTC software.

serb1231, 2019-11-18:
The robot controller Android device has the 5.0 version .Should I downgrade
or upgrade this version?

dum., 17 nov. 2019, 8:06 p.m. Noah Andrews <notifications@github.com> a
scris:

gearsincorg, 2019-11-18:
> The robot controller Android device has the 5.0 version

That sounds right.. Android Version 5.x was not an allowed, mainly because there were issues with it only seeing one USB controller.

The allowed versions are listed below (from Game manual 1. Section 7.3.4 Robot Software Rules)

a. ....  4.4 or higher (Kit Kat)
b. ....  7.0 or higher (Nougat)
c. ....  6.0 or higher (Marshmallow) 

If possible you should upgrade to the most recent version, available for your phones (and on the list above)

serb1231, 2019-11-18:
Thank you very much for your advice

lun., 18 nov. 2019, 3:10 p.m. Phil Malone <notifications@github.com> a
scris:

closed 2019-11-29 by serb1231