ftctechnh/ftc_app #743: Error in code is pointing to non existent source files

Zach-coder, 2019-11-24:


cmacfarl, 2019-11-24:
Do you have build logsm and can you post your software somewhere public (github would be ideal)?

Zach-coder, 2019-11-27:
I have a repository set up on github.
Repository name is 9507-RD-Code

cmacfarl, 2019-11-28:
Github search has no results for 9507-RD-Code.  Can you post the full url to the repository?

Also build logs would be helpful.

Note that sometimes Android Studio goes off in the weeds and gives build errors that are not related to FTC software.  When that happens I typically do File -> Invalidate Caches and Restart to fix the problem.

Zach-coder, 2019-12-01:
Thank you. At this point I fixed the problem. I removed the old FTC source code and reinstalled this years version. I think I missed one of the old source code files. I wiped android studio and re-downloaded the new source files.

closed 2019-12-01 by cmacfarl