ftctechnh/ftc_app #747: I can't press the Init Button

serb1231, 2019-11-30:
We have 2 Samsung Galaxy S5 phones(android version 6.1).We have connected the RC to the Robot(witch has 4 motors and 2 servo-motors),it is detecting the components,the phones are connected through wifi direct,but when we try to press the Init Button,we can see the play button for a fraction of a second but then it goes back to the init button.

gearsincorg, 2019-11-30:
One possible reason is that your opmode is just running and then stopping because your program has ended.

We need more info.  What opmode are you running? can you paste a copy of it here (using the Insert Code feature at the top of the edit box).

gearsincorg, 2019-11-30:
BTW, this is the OLD github repository.  You should be downloading code and posting issues here:

https://github.com/FIRST-Tech-Challenge/SkyStone

serb1231, 2019-11-30:
```
package org.firstinspires.ftc.teamcode;

import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;
import com.qualcomm.robotcore.eventloop.opmode.TeleOp;
import com.qualcomm.robotcore.hardware.DcMotor;
import `com.qualcomm.robotcore.hardware.Servo;`

/**
 * This OpMode ramps a single motor speed up and down repeatedly until Stop is pressed.
 * The code is structured as a LinearOpMode
 *
 * This code assumes a DC motor configured with the name "left_drive" as is found on a pushbot.
 *
 * INCREMENT sets how much to increase/decrease the power each cycle
 * CYCLE_MS sets the update period.
 *
 * Use Android Studio to Copy this Class, and Paste it into your team's code folder with a new name.
 * Remove or comment out the @Disabled line to add this opmode to the Driver Station OpMode list
 */
@TeleOp(name = "Concept: Ramp Motor Speed", group = "Concept")
//@Disabled
public class TestConceptRampMotorSpeed extends LinearOpMode {

    static final double INCREMENT   = 0.01;     // amount to ramp motor each CYCLE_MS cycle
    static final int    CYCLE_MS    =   50;     // period of each cycle
    static final double MAX_FWD     =  1.0;     // Maximum FWD power applied to motor
    static final double MAX_REV     = -1.0;     // Maximum REV power applied to motor

    // Define class members
    DcMotor motor1;
    DcMotor motor2;
    DcMotor motor3;
    DcMotor motor4;
    DcMotor motor5;
    double  power   = 0;
    boolean rampUp  = true;

    static final double MAX_POS     =  1.0;     // Maximum rotational position
    static final double MIN_POS     =  0.0;     // Minimum rotational position
    static final double increment=0.01;

    Servo servo1;
    //Servo servo2;
    double  position = (MAX_POS - MIN_POS) / 2; // Start at halfway position

    @Override
    public void runOpMode() {
    

        // Connect to motor (Assume standard left wheel)
        // Change the text in quotes to match any motor name on your robot.
        motor1 = hardwareMap.get(DcMotor.class, "left_front_drive");
        motor2 = hardwareMap.get(DcMotor.class, "right_front_drive");
        motor3 = hardwareMap.get(DcMotor.class, "left_back_drive");
        motor4 = hardwareMap.get(DcMotor.class, "right_back_drive");
        

        while(opModeIsActive()) {  

            rampUp=true;
            // Ramp the motors, according to the rampUp variable.
            if (rampUp) {
                // Keep stepping up until we hit the max value.
                power += INCREMENT ;
                if (power >= MAX_FWD ) {
                    power = MAX_FWD;
                    rampUp = !rampUp;   // Switch ramp direction
                }
            }
            else {
                // Keep stepping down until we hit the min value.
                power -= INCREMENT ;
                if (power <= MAX_REV ) {
                    power = MAX_REV;
                    rampUp = !rampUp;  // Switch ramp direction
                }
            }

    rampUp=true;

            servo1 = hardwareMap.get(Servo.class, "left_hand");
            //servo2 = hardwareMap.get(Servo.class, "right_hand");

                if (rampUp) {
                    // Keep stepping up until we hit the max value.
                    position += increment ;
                    if (position >= MAX_POS ) {
                        position = MAX_POS;
                        rampUp = !rampUp;   // Switch ramp direction
                    }
                }
                else {
                    // Keep stepping down until we hit the min value.
                    position -= increment ;
                    if (position <= MIN_POS ) {
                        position = MIN_POS;
                        rampUp = !rampUp;  // Switch ramp direction
                    }
                }

                servo1.setPosition(position);
                //servo2.setPosition(position);

            motor1.setPower(power);
            motor2.setPower(-power);
            motor3.setPower(power);
            motor4.setPower(-power);
            sleep(CYCLE_MS);
            idle();
        }

        // Turn off motor and signal done;
        motor1.setPower(0);
        motor2.setPower(0);
        motor3.setPower(0);
        motor4.setPower(0);
        telemetry.addData(">", "Done");
        telemetry.update();

    }
}
```

serb1231, 2019-11-30:
Thank you.I will post it on the new repository.The program works well until the While Loop.It wont enter the while loop.

cmacfarl, 2019-11-30:
You need a call to waitForStart().

Take a look at any of the Concept opmodes in external/samples for examples of where to put it.

serb1231, 2019-11-30:
Thank you @cmacfarl

gearsincorg, 2019-12-01:
More Info:

So, the reason that the opmode exited immediately is that opModeIsActive() is not set to true until after Play is pressed, so since you did not do a waitForStart(), the opmode immediately started the while loop.  And since opModeIsActive() returned false, the loop exited immediately.

Also since you have combined the motor ramp and servo sweep examples, you really need two different rampUp variables.  Use one for the servo, and one for the motor.

eg:  
boolean rampUpServo = false;
boolean rampUpMotor = false;

serb1231, 2019-12-01:
Thank you @gearsincorg ! I understand what i need to do now.

closed 2019-12-01 by cmacfarl