ftctechnh/ftc_app #749: RC stuck in initialization  w/ warning contains RUN_TO_POSITION

sniper-feng, 2019-12-07:
Our group has got several windows users while I am using macOS Version of Android Studio.As we are switching between the 2 OS,RobotController always stuck in initialization period.It also alternates with several different kind of warnings below.
1,"You must set target position before using RUN_TO_POSITION"
2."getCurrentPosition() on a null referenced object"

These problem sometimes can be solved by rebooting the phone ,installing the RC app and configure the configuration once again but it really annoys.

cmacfarl, 2019-12-07:
The first message is telling you what the problem is.  If you are going to set the runMode to RUN_TO_POSITION, then you must also set a target position via a call to setTargetPosition() otherwise how does it know what position to run to?

As to the null pointer error, we'd need to see actual software.  If you have a github repo you could reference here that would be ideal.

sniper-feng, 2019-12-29:
> The first message is telling you what the problem is. If you are going to set the runMode to RUN_TO_POSITION, then you must also set a target position via a call to setTargetPosition() otherwise how [...]

sniper-feng, 2019-12-29:
The problem is I searched the usage of  the `RUN_TO_POSITION` enumeration but find nothing. Then, the problem still exists, but unexpectedly we solved this problem by  switching to another expansion hub.But we need to know if this problem is connected with a particular expansion hub.

As for the null pointer error, we have already know that the RC app will ignore the declaration outside the `runOpMode()` method call.

All in all, thanks for your help

FROGbots-4634, 2019-12-29:
I suppose it's technically possible there could be a firmware bug that could cause RTP mode to persist through the failsafe commands sent by the StopRobot OpMode. I might do some testing later and see if I can reproduce.