ftctechnh/ftc_app #750: Robot Controller Will Not Connect To REV Expansion Hub

ValleyInnovators, 2019-12-10:
We've been trying to get our phone set up to finally test our code but they just won't connect to the REV Expansion Hub. We've tried swapping cables, reinstalling the app, and restarting the phone and robot but with no luck. When we plug in the phone, the light on the hub doesn't even light up. Any ideas on what may be the issue?

FROGbots-4634, 2019-12-10:
*"When we plug in the phone, the light on the hub doesn't even light up."* - does it light up when you plug it into a computer? If not, it seems you have a fried ExH.

gearsincorg, 2019-12-11:
If you have a motorola phone (and probably others), the Hub won't light up unless you have 12V connected.

Have you connected it to a 12V battery?

ps: For technical help, the place to post is here:
https://ftcforum.firstinspires.org/forum/ftc-technology

ValleyInnovators, 2019-12-11:
We have connected it to a 12V battery with power but the phone doesn't connect to the Hub for some reason. Also, the phone doesn't automatically open the Robot Controller App when we plug it in and it doesn't give us the option to make the phone open the app by default.

NoahAndrews, 2019-12-11:
Sounds like a bad hub. Email support@revrobotics.com

AlecHub, 2019-12-11:
> If you have a motorola phone (and probably others), the Hub won't light up unless you have 12V connected.

When we connect our Expansion Hubs to a Motorola G2, G3, or G5 without the 12V battery connected, the LED on the hubs flash orange.  Also, if the two hubs are connected via only the 12V ports (XT30 port to XT30 port), both hubs flash orange even though there is no phone connected to the second.  It seems the first hub bleeds 6V to the second hub via the 12V port.

gearsincorg, 2019-12-12:
@AlecHub   You are confusing the issue somewhat.

The Moto G phones only power the Expansion hubs under some conditions.  They won't automatically power the Hubs if they don't have 12V connected when the phones are first connected.  However, once the 12V has been applied, the phone will continue to power them with 5V if the 12V is removed..

Background...  The ZTE phones would always power the hubs but the Moto Gs are different.  If I plug my G2 or G4 into an unpowered hub, the App will not auto launch (or provide 5V).  If I turn on the 12V, the phone seems to be happy and then the App will launch.  Now if I remove 12V the phone powers the hub and the LED flashes orrange.

 @ValleyInnovators said that when they plug the phones in the hub does not light up....  This can be normal on Moto G phones, so I wanted to make sure the HUB was powered.

AlecHub, 2019-12-12:
> @AlecHub   You are confusing the issue somewhat.
>
> The Moto G phones only power the Expansion hubs under some conditions.  They won't automatically power the Hubs if they don't have 12V connected [...]

When we connect our [legal] Moto Gx phones to the expansion hub, with nothing else connected to the hub, the RC app launches on the Moto G phone and the status LED of the hub starts blinking orange.

gearsincorg, 2019-12-12:
> When we connect our [legal] Moto Gx phones to the expansion hub, with nothing else connected to the hub, the RC app launches on the Moto G phone and the status LED of the hub starts blinking orange.

That's great...   I'm just saying that this is not "universal behavior".  

It could be REV EH or Phone OS version specific.   I don't know, but I started observing the alternate behavior last season while CSA'ing at events.

It can be a point of confusion if a team's phones do this.  This is why I ALWAYS recommend turning the 12V power on FIRST, before plugging in the phone and expecting an auto-launch.