ftctechnh/ftc_app #751: Webcam Consistently Crashing on Init

sjpeccoud, 2019-12-14:
During my team's autonomous we use a logitech webcam runnng the tfod  for Skystone. It initializes ok 1/5 times, usually it is a tedious process of unplugging and plugging the webcam. Does anyone have any suggestions?

FROGbots-4634, 2019-12-15:
Can you post your logs from a failed attempt?

sjpeccoud, 2019-12-15:
The console log crashes because of a getFocalLength() method but I believe that the phone simply cannot see the webcam most of the time on init.

cmacfarl, 2019-12-15:
The entire stack trace would be helpful.

sjpeccoud, 2019-12-15:
Ok, I will have on Monday.

sjpeccoud, 2019-12-16:
So here is the full error message:

Robot Status: EMERGENCY STOP
To attempt recovery, please restart the robot.
Error: User Code threw an uncaught exception: NullPointerException - Attempt to invoke virtual method 'com.vuforia.Vec2F
com.vuforia.CameraCalibration.getFocalLength()' on a null object reference.

FROGbots-4634, 2019-12-16:
@sjpeccoud that is not the entire stack trace, that's only like the first line or two

sjpeccoud, 2019-12-16:
Thats all there was on the phone screen. Where can I find the whole message?

FROGbots-4634, 2019-12-17:
You can pull the `robotControllerLog.txt` file from the `FIRST` folder on the internal storage of the phone and upload it here for us to look at.

KotoTheOriginal, 2019-12-19:
Hey, I'm on the same team as Sam and am looking into finding out the reasoning for this error. The closest I could find to what you mentioned about finding the whole error message was looking at the logs on the RC phone. How can I find the internal storage of the phone to find this FIRST folder?

cmacfarl, 2019-12-19:
adb and familiarity with a shell is the easiest way.  They are in /sdcard/ and named robotControllerLog.txt, rotated files have numbers appended to the filename.

adb details can be found here: https://developer.android.com/studio/command-line/adb

If you don't want to deal that, you can select View Logs, and if you long-click on the logs that appear the standard Android sharing dialog will appear whence you can choose how you'd like to share them.

KokeCacao, 2020-01-03:
> Hey, I'm on the same team as Sam and am looking into finding out the reasoning for this error. The closest I could find to what you mentioned about finding the whole error message was looking at the logs on the RC phone. How can I find the internal storage of the phone to find this FIRST folder?

for ref:
do `adb pull /sdcard/robotControllerLog.txt`