ftctechnh/ftc_app #754: LynxModule retransmitting happens randomly in autonomous run, block TeamCode from execution half second to 2 seconds.

kzhou57, 2020-01-22:
`
01-21 16:58:38.442  2639  2989 D RobotCore: DRIVING! Distance: 44.9
01-21 16:58:38.580  2639  2925 V LynxModule: retransmitting: mod=2 cmd=0x7f04 msg#=10 ref#=0 
01-21 16:58:38.683  2639  2925 V LynxModule: retransmitting: mod=2 cmd=0x7f04 msg#=10 ref#=0 
01-21 16:58:38.785  2639  2925 V LynxModule: retransmitting: mod=2 cmd=0x7f04 msg#=10 ref#=0 
01-21 16:58:38.887  2639  2925 V LynxModule: retransmitting: mod=2 cmd=0x7f04 msg#=10 ref#=0 
01-21 16:58:38.980  2639  2925 V LynxModule: retransmitting: mod=2 cmd=0x7f04 msg#=10 ref#=0 
01-21 16:58:38.984  2639  2925 V RobotCore: timeout: abandoning waiting 500ms for ack: cmd=LynxKeepAliveCommand mod=2 msg#=10
01-21 16:58:38.983  2639  2989 E RobotCore: lynx xmit lock: #### abandoning lock: old=LynxKeepAliveCommand(10)
01-21 16:58:38.984  2639  2925 V RobotCore: nack rec'd mod=2 msg#=10 ref#=0 reason=ABANDONED_WAITING_FOR_ACK:257
01-21 16:58:38.984  2639  2989 E RobotCore: lynx xmit lock:                       new=LynxI2cReadStatusQueryCommand(0)
01-21 16:58:38.985  2639  2925 E RobotCore: lynx xmit lock: #### incorrect owner releasing message keyed lock: ignored: old=LynxI2cReadStatusQueryCommand(1:252)
01-21 16:58:38.985  2639  2925 E RobotCore: lynx xmit lock:                                                             new=LynxKeepAliveCommand(2:10)
01-21 16:58:38.985  2639  2925 W LynxModule: LynxKeepAliveCommand was abandoned waiting for ack
01-21 16:58:39.071  2639  2989 D RobotCore: DRIVING! Distance: 32.5
`

FROGbots-4634, 2020-01-22:
Hmm, if it were an issue with the slave Lynx, I would say replace your RS485 cable. But since it's the master Lynx (you're only using one) I would say replace your OTG cable, replace your mini USB cable, and double check to make sure your 12v power isn't "bliping".

Does this only happen in auto, or during teleop as well? Also, from the logs it appears the Lynx stopped responding during an I2C operation; have you seen this error in other circumstances?

kzhou57, 2020-01-22:
It seems the controller phone (or it's USB port) is the problem. We flipped the controller phone and driver phone (both them are stock Moto G4 play ), and didn't see the error again any . 

Can we make this error more obviously ? For example, show something on the telemetry ? It took us long time to investigate.

FROGbots-4634, 2020-01-22:
SDK v5.4 includes improved logging that I believe covers this scenario

kzhou57, 2020-01-22:
Excellent ! Thank you very much, I am going to close the ticket.

closed 2020-01-22 by kzhou57